nathanmarlor/foxess_modbus #10: Adding Intergration Error

Chrisbrown01942, 2023-03-08:
Hi Nathan, followed the usual HACS procedure to download foxess_modbus, and all is fine. But when I try to add the integration I receive an error & can not add it.

[image]
[image]
 

## Version of the custom_component v1.0.0

## Configuration

```yaml
Add your logs here.
```

## Describe the bug

A clear and concise description of what the bug is.

## Debug log

```text

Add your logs here.

```

nathanmarlor, 2023-03-08:
Can you add some logs please?

Chrisbrown01942, 2023-03-08:
To enable a logger is it as simple as adding the below?
# Example configuration.yaml entry
logger:

nathanmarlor, 2023-03-08:
If it's failing to setup there should be something significant in the system -> logs.

Chrisbrown01942, 2023-03-08:
ok got it
Logger: homeassistant.config_entries
Source: config_entries.py:834
First occurred: 18:24:06 (2 occurrences)
Last logged: 18:24:06

Error occurred loading configuration flow for integration foxess_modbus: cannot import name 'ModbusTcpClient' from 'pymodbus.client' (/usr/local/lib/python3.10/site-packages/pymodbus/client/__init__.py)

nathanmarlor, 2023-03-08:
What version of HA are you using please? Did you restart after adding the component through HACS?

Chrisbrown01942, 2023-03-08:
Home Assistant 2022.12.7
Supervisor 2023.03.1
Operating System 9.4
Frontend 20221213.1 - latest

Yes, I restarted HA after downloading via HACS

nathanmarlor, 2023-03-08:
I might have overlooked something silly.

Can you try -> https://github.com/nathanmarlor/foxess_modbus/releases/tag/v1.0.1 please?

Chrisbrown01942, 2023-03-08:
Sorry, didn't work. I've added the new log for the issue.

Logger: homeassistant.config_entries
Source: config_entries.py:834
First occurred: 18:48:48 (1 occurrences)
Last logged: 18:48:48

Error occurred loading configuration flow for integration foxess_modbus: cannot import name 'ModbusTcpClient' from 'pymodbus.client' (/usr/local/lib/python3.10/site-packages/pymodbus/client/__init__.py)

Chrisbrown01942, 2023-03-08:
I do have a HA core & OS update avaible. Should I do that and get back to you?

nathanmarlor, 2023-03-08:
Yes please!

nathanmarlor, 2023-03-08:
Also worth trying v1.0.3 - I think it's related to an out of date pymodbus version in your HA. (ex: https://github.com/binsentsu/home-assistant-solaredge-modbus/issues/143)

Chrisbrown01942, 2023-03-08:
Tried v1.0.3 same issue. I'll update HA & get back to you.

Logger: homeassistant.config_entries
Source: config_entries.py:834
First occurred: 19:34:25 (1 occurrences)
Last logged: 19:34:25

Error occurred loading configuration flow for integration foxess_modbus: cannot import name 'ModbusTcpClient' from 'pymodbus.client' (/usr/local/lib/python3.10/site-packages/pymodbus/client/__init__.py)

Chrisbrown01942, 2023-03-08:
Updated HA to the latest version & now have it configured & working
Home Assistant 2023.3.2
Supervisor 2023.03.1
Operating System 9.5
Frontend 20230306.0 - latest

nathanmarlor, 2023-03-08:
Wahhhooo! :D

closed 2023-03-08 by nathanmarlor