nathanmarlor/foxess_modbus #11: Error connecting Integration

andykorol1, 2023-03-08:
[image]

I am trying to add the integration and when I put the IP address in it throws the fault in the screenshot. Anything I am doing wrong?

nathanmarlor, 2023-03-08:
Can you include some logs please?

andykorol1, 2023-03-08:
I can't see anything in the logs in regards to this integration. I have removed the original foxess modbus integration. Including deleting all the config.yaml section of it too

nathanmarlor, 2023-03-08:
Are you using the same IP? Same port? H1 inverter? Slave address 247? LAN or W610 (or other RTU to TCP adapter)?

andykorol1, 2023-03-08:
Same IP as before, same port 502, H1 inverter, Slave should still be 247 and it is hard wired via LAN. 
Master version 1.48
Slave version 1.02
Manager version 1.45

nathanmarlor, 2023-03-08:
Ok, can you try -> v1.0.2 please and let me know what's in the logs? You'll need to turn on debug logging for the component.

andykorol1, 2023-03-08:
On version 1.0.3 now and still the same. Here is the only thing I have seen in the logs for it.

Logger: homeassistant.components.modbus.modbus
Source: components/modbus/modbus.py:370
Integration: Modbus (documentation, issues)
First occurred: 7:31:37 PM (1 occurrences)
Last logged: 7:31:37 PM

modbus HA-FoxESS-Modbus-LAN communication closed

nathanmarlor, 2023-03-08:
Can you send me a screenshot of that log message please? I don't recognise the format.

andykorol1, 2023-03-08:
Here is another log I can get if input an incorrect IP address but it doesn't appear when I put the correct IP in.

This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/foxess_modbus/modbus_client.py:22
Integration: FoxESS - Modbus (documentation, issues)
First occurred: 7:38:01 PM (2 occurrences)
Last logged: 7:38:04 PM

Connection to (192.168.1.211, 502) failed: timed out

nathanmarlor, 2023-03-08:
Have you disabled the other modbus integration? The previous log you sent 'connection closed' looks to be from the HA modbus client.

nathanmarlor, 2023-03-08:
Just comment out the references in your configuration file.

If you hit 'full logs' do you see this? It's set at info level so it won't appear in the overview window.
 https://github.com/nathanmarlor/foxess_modbus/blob/main/custom_components/foxess_modbus/modbus_client.py#L30

andykorol1, 2023-03-08:
I don't see what you've sent me but could you clarify what I need in my config file to get the logging you require?

nathanmarlor, 2023-03-08:
Add this:

logger:
  default: info
  logs:
    custom_components.foxess_modbus: debug

andykorol1, 2023-03-08:
[image]

I get this by adding that?

nathanmarlor, 2023-03-09:
It mustn't register the logger before the config has completed.

The only thing I can suggest is to try modpoll directly ->https://www.modbusdriver.com/modpoll.html

andykorol1, 2023-03-10:
Logger: pymodbus.logging
Source: runner.py:179
First occurred: 6:32:34 PM (2 occurrences)
Last logged: 6:32:38 PM

Ignoring launch of delayed reconnection, another is in progress
Failed to connect: [Errno 104] Connect call failed ('192.168.1.22', 502)

This is the only error I can get and still can't get it to connect with 1.0.4

andykorol1, 2023-03-10:
This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/foxess_modbus/modbus_client.py:22
Integration: FoxESS - Modbus (documentation, issues)
First occurred: 6:50:27 PM (3 occurrences)
Last logged: 6:50:27 PM

Failed to connect: [Errno 111] Connect call failed ('192.168.1.1', 502)
Ignoring launch of delayed reconnection, another is in progress

I can get this error if I put an incorrect IP in

nathanmarlor, 2023-03-10:
Are you sure you don't have the original modbus integration running? The LAN port only seems to like one connection.

andykorol1, 2023-03-11:
I have deleted all the information from the config file and anything in the custom components folder too. Also removed anything from HACS. Restarted and still the same. But I do feel like something on the network is still talking to the Modbus port or it could be something to do with my home network potentially stopping this integration from having access. How much does this differ to the original Modbus integration?

nathanmarlor, 2023-03-11:
I haven't checked but it's likely using the exact same pymodbus library. I've been testing against both the W610 and LAN port and noticed that the LAN port stopped responding. I forced a disconnect and it came back alive shortly after. Maybe worth disconnecting your ethernet connection, letting the inverter lose it's IP etc then reconnect?

andykorol1, 2023-03-11:
I have set my inverter with a fixed IP so I might change it to another address and cycle the connection to see what it does.

andykorol1, 2023-03-11:
[image]

Very new to modpoll but here is what I am able to get so far.

benbeton, 2023-03-11:
I do have the same issue with my USR-610. there is no other integration running that can cause the connection error on modbus.
When using [gmodbus](https://www.gineers.com/en/products_soft_other_gmodbus.php), I can connect perfectly fine and read data
restarting the USR-610 does not help

andykorol1, 2023-03-11:
I wonder if this is a firmware issue at the inverter end? I am on Master version 1.48 Slave version 1.02 Manager version 1.45

benbeton, 2023-03-11:
I'm don't think so, because I can read all the data from the USR-610 
but just to check, I have a **H3**: 
**Master : 1.67 Slave : 1.02Manager : 1.51**

nathanmarlor, 2023-03-12:
@benbeton H3s aren't supported yet.

benbeton, 2023-03-14:
@nathanmarlor I know, but I thought the USR should be usable aswell, maybe with restrictions. I'll happy to help with the H3 integration

andykorol1, 2023-03-15:
This error originated from a custom integration.

Logger: custom_components.foxess_modbus.config_flow
Source: custom_components/foxess_modbus/config_flow.py:115
Integration: FoxESS - Modbus (documentation, issues)
First occurred: 8:14:31 PM (1 occurrences)
Last logged: 8:14:31 PM

ModbusIOException()

Here is the latest error i can see

nathanmarlor, 2023-03-15:
@andykorol1 it just can't connect to your device for some reason - have you tried modpoll?

andykorol1, 2023-03-15:
[image]

This is all I seem to get using modpoll, do you think my home network could have influence in it not working?

nathanmarlor, 2023-03-15:
@andykorol1 can you poll address 10000 and 10001?

andykorol1, 2023-03-15:
[image]

Is this the correct way to do it?

nathanmarlor, 2023-03-15:
Yep, looks to be. Next suggestion would be to try power cycling the inverter? I had to do that once during testing, it got itself into a weird state.

andykorol1, 2023-03-15:
I've done a full power cycle of the inverter and the same issue. I can ping the IP address from my PC fine. 

Network wise my inverter is plugged into a switch which also has the HA in it and then I have a Unifi Dream router which the PC is connected to. I find it really strange the old Modbus integration works fine. Cheers for the help

nathanmarlor, 2023-03-15:
Is the old modbus integration definitely removed? Try turning HA off completely then hitting modpoll?

andykorol1, 2023-03-15:
Did the full reboot and nothing changed. Just to clarify this can be done via the direct LAN port on the inverter?

nathanmarlor, 2023-03-15:
Yep, I've been testing against both RTU -> TCP and direct into the LAN port. Copy/paste your HA config here and I'll have a quick look?

andykorol1, 2023-03-15:
`
# Loads default set of integrations. Do not remove.
default_config:

# Text to speech
tts:
  - platform: google_translate

automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml
frontend:
  themes: !include_dir_merge_named themes

# FoxESS Modbus Plugin - Configuration and Sensor Mappings
# Source https://github.com/StealthChesnut/HA-FoxESS-Modbus
# This file isn't a valid configuration file!

# Instructions:
# copy the custom_components directory to your HA custom_components directory
# Add ONE of the modbus: lines to your configuration.yaml
# Add the sensors below to your configuration.yaml 
#
# USB: Edit the file to update the connection settings
#
# LAN: Add a new secret to your secrets file - FoxESSInverterIP: <your inverter IP>

#modbus: !include custom_components/HA-FoxESS-Modbus/modbusUSB.yaml
# modbus: !include custom_components/HA-FoxESS-Modbus/modbusLAN.yaml

# #Sensors turn power values into power/hour values for meters
# sensor: 
#   - method: left
#     name: pv1_sum
#     platform: integration
#     round: 2
#     source: sensor.pv1_power
#     unit_time: h
#   - method: left
#     name: pv2_sum
#     platform: integration
#     round: 2
#     source: sensor.pv2_power
#     unit_time: h
#   - method: left
#     name: pv_total_sum
#     platform: integration
#     round: 2
#     source: sensor.pv_power_now
#     unit_time: h
#   - method: left
#     name: load_sum
#     platform: integration
#     round: 2
#     source: sensor.load_power
#     unit_time: h
#   - method: left
#     name: bat_charge_sum
#     platform: integration
#     round: 2
#     source: sensor.battery_charge
#     unit_time: h
#   - method: left
#     name: bat_discharge_sum
#     platform: integration
#     round: 2
#     source: sensor.battery_discharge
#     unit_time: h
#   - method: left
#     name: feedin_sum
[... 107 lines cut ...]
# utility_meter:
#   load_daily:
#     source: sensor.load_sum
#     cycle: daily
#   pv1_daily:
#     source: sensor.pv1_sum
#     cycle: daily
#   pv2_daily:
#     source: sensor.pv2_sum
#     cycle: daily
#   bat_charge_daily:
#     source: sensor.bat_charge_sum
#     cycle: daily
#   bat_discharge_daily:
#     source: sensor.bat_discharge_sum
#     cycle: daily
#   feedin_daily:
#     source: sensor.feedin_sum
#     cycle: daily
#   grid_daily:
#     source: sensor.grid_consumption_sum
#     cycle: daily
#   losses_daily:
#     source: sensor.losses_sum
#     cycle: daily`

andykorol1, 2023-03-15:
# Loads default set of integrations. Do not remove.
default_config:

# Text to speech
tts:
  - platform: google_translate

automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml
frontend:
  themes: !include_dir_merge_named themes

# FoxESS Modbus Plugin - Configuration and Sensor Mappings
# Source https://github.com/StealthChesnut/HA-FoxESS-Modbus
# This file isn't a valid configuration file!

# Instructions:
# copy the custom_components directory to your HA custom_components directory
# Add ONE of the modbus: lines to your configuration.yaml
# Add the sensors below to your configuration.yaml 
#
# USB: Edit the file to update the connection settings
#
# LAN: Add a new secret to your secrets file - FoxESSInverterIP: <your inverter IP>

#modbus: !include custom_components/HA-FoxESS-Modbus/modbusUSB.yaml
# modbus: !include custom_components/HA-FoxESS-Modbus/modbusLAN.yaml

# #Sensors turn power values into power/hour values for meters
# sensor: 
#   - method: left
#     name: pv1_sum
#     platform: integration
#     round: 2
#     source: sensor.pv1_power
#     unit_time: h
#   - method: left
#     name: pv2_sum
#     platform: integration
#     round: 2
#     source: sensor.pv2_power
#     unit_time: h
#   - method: left
#     name: pv_total_sum
#     platform: integration
#     round: 2
#     source: sensor.pv_power_now
#     unit_time: h
#   - method: left
#     name: load_sum
#     platform: integration
#     round: 2
#     source: sensor.load_power
#     unit_time: h
#   - method: left
#     name: bat_charge_sum
#     platform: integration
#     round: 2
#     source: sensor.battery_charge
#     unit_time: h
#   - method: left
#     name: bat_discharge_sum
#     platform: integration
#     round: 2
#     source: sensor.battery_discharge
#     unit_time: h
#   - method: left
#     name: feedin_sum
#     platform: integration
[... 106 lines cut ...]
# utility_meter:
#   load_daily:
#     source: sensor.load_sum
#     cycle: daily
#   pv1_daily:
#     source: sensor.pv1_sum
#     cycle: daily
#   pv2_daily:
#     source: sensor.pv2_sum
#     cycle: daily
#   bat_charge_daily:
#     source: sensor.bat_charge_sum
#     cycle: daily
#   bat_discharge_daily:
#     source: sensor.bat_discharge_sum
#     cycle: daily
#   feedin_daily:
#     source: sensor.feedin_sum
#     cycle: daily
#   grid_daily:
#     source: sensor.grid_consumption_sum
#     cycle: daily
#   losses_daily:
#     source: sensor.losses_sum
#     cycle: daily

andykorol1, 2023-03-15:
All my comments are missing when I paste it?

nathanmarlor, 2023-03-15:
@andykorol1  you've still got the modbus include in there?

andykorol1, 2023-03-15:
[image]

nathanmarlor, 2023-03-15:
So you reboot with that, confirm no sensors load etc. but modpoll still doesn't work?

andykorol1, 2023-03-15:
Yeah reboot with it as is, my sensors still show in my dashboard but just at the last known value as they still show up as an entity in the entities list. Had another play with modpoll can get it to show a reply time out error now by changing it to RTU over TCP

andykorol1, 2023-03-15:
I'm definitely seeing some thing from the inverter via modpoll as if I put an incorrect IP in it says there is no slave but with the correct IP I get the reply time out error. What version of firmware are you using it on?

nathanmarlor, 2023-03-15:
That could be it - perhaps they put the model type in a different place in the modbus address space. Can you get anything from addresses 10000 or 30000? That's where the model number for RTU and TCP connections lives respectively.

andykorol1, 2023-03-15:
I've been trying to use CAS modbus and get the following error
TCP 192.168.1.23:502 timeout: 3
	Device: 247
		Read Coil status starting at 30000 for 1

[03/15/23 22:16:28] Ready...
[03/15/23 22:16:49] Connect to TCP 192.168.1.23:502 timeout: 3
[03/15/23 22:16:49] => Poll: F7 01 01 2B 00 01 
[03/15/23 22:16:49] <= Response: F7 81 02 
[03/15/23 22:17:21] Recv error: Exception Response:
The device responded but with an error instead of data

0x02 - ILLEGAL DATA ADDRESS

andykorol1, 2023-03-15:
Progress finally!!
[image]

nathanmarlor, 2023-03-16:
@andykorol1 your model number starts with AIO so the autodetect isn't working. Should be able to add it. To confirm you were using the modbus USB file with no modifications?

andykorol1, 2023-03-16:
Modbus LAN with no modifications

nathanmarlor, 2023-03-16:
Can you try this please?

https://github.com/nathanmarlor/foxess_modbus/releases/tag/v1.3.0b1

andykorol1, 2023-03-16:
I've got the same issue still, but I am doing it remotely on my phone so will need to go through it all when I get home

m24andrew, 2023-03-19:
I too have an AIO that is connected via LAN and worked previously.  That instance of HA is shut down and I am now testing on another.  I have the latest version and reset everything as described above.  I get the same error as the very first post after I enter my modbus host.
I get the same output from modpoll as @andykorol1 just above with the same command and below is my log:

`2023-03-19 13:17:23.233 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (10000, 10, (247))
2023-03-19 13:17:23.233 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Connecting to modbus - ({'modbus_type': 'TCP', 'host': '192.168.10.101', 'port': '502'})
2023-03-19 13:17:23.300 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Closing connection to modbus
2023-03-19 13:17:23.301 WARNING (MainThread) [custom_components.foxess_modbus.config_flow] ModbusIOException()
2023-03-19 13:19:08.000 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (10000, 10, (247))
2023-03-19 13:19:08.000 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Connecting to modbus - ({'modbus_type': 'TCP', 'host': '192.168.10.101', 'port': '502'})
2023-03-19 13:19:08.087 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Closing connection to modbus
2023-03-19 13:19:08.088 WARNING (MainThread) [custom_components.foxess_modbus.config_flow] ModbusIOException()
2023-03-19 13:45:16.118 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (10000, 10, (247))
2023-03-19 13:45:16.118 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Connecting to modbus - ({'modbus_type': 'TCP', 'host': '192.168.10.101', 'port': '502'})
2023-03-19 13:45:16.343 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Closing connection to modbus
2023-03-19 13:45:16.344 WARNING (MainThread) [custom_components.foxess_modbus.config_flow] ModbusIOException()`

nathanmarlor, 2023-03-20:
@andykorol1 @m24andrew please try v1.4.0b2.

Ndrinta, 2023-03-20:
Hi Nathan, unfortunately I am experiencing the same here.
To exclude all possibile issues due to my HA installation I stopped the VM and created a brand new VM_test (then installed HACS and downloaded your script, nothing else).
I get the same error as the other guys here, but if I open modpoll from the same machine I can get a response just fine.

[image]

I'm reaching 30000 because I'm still running on pure LAN.
I'm sorry if I'm giving you headaches, it's really not my intention but something is weird here (and as you said it's likely my/our fault but I can't really pinpoint what the issue is...)

Set-up:
PC (running VM*) (and running modpoll) <--> Router <--> Unmanaged Switch <--> H1 Inverter [LAN port]

VM* fresh ova_9.5 downloaded tonight

Questions:
Do you see anything wrong with my setup, or that you'd like to see different?
Did you try connecting with pure LAN to your Inverter? Or are you aware of anyone who successfully connected in this way?

Ndrinta, 2023-03-20:
Tomorrow morning I'm gonna start again by trying to restart the inverter. Maybe that's it... I really DK

Ah BTW, just to leave a trace:
Master : 1.54
Slave : 1.02
Manager : 1.57

nathanmarlor, 2023-03-20:
@Ndrinta Have you tried the latest bet?

Ndrinta, 2023-03-20:
On it.

Ndrinta, 2023-03-20:
Winner winner chicken dinner!

[image]

nathanmarlor, 2023-03-20:
Hopefully @m24andrew and @andykorol1 have similar success!

Ndrinta, 2023-03-20:
I'm now comparing the newly created entities with stuff I might have altered already in my config because I found a few "_2", and I know that should't have happened.

[3 more replies]

closed 2023-03-20 by nathanmarlor