nathanmarlor/foxess_modbus #6: Add H3 Inverter Sensors

nathanmarlor, 2023-03-08:


benbeton, 2023-03-14:
I'm happy to help, let me know what you need for the H3 inverter support

nathanmarlor, 2023-03-14:
Thanks @benbeton 

Basically need one of these creating for H3 aux and LAN -> https://github.com/nathanmarlor/foxess_modbus/blob/main/custom_components/foxess_modbus/sensors/h1_aux_sensors.py

I can do the rest :)

canton7, 2023-03-17:
Permalink: https://github.com/nathanmarlor/foxess_modbus/blob/b519e72318512df877490320232e1f1a354a5e20/custom_components/foxess_modbus/entities/xx1_aux_sensors.py

benbeton, 2023-03-18:
created the first sensors: https://github.com/nathanmarlor/foxess_modbus/pull/57
Source for H3 Sensors: https://github.com/rsaemann/HA-FoxESS-H3-Modbus/blob/main/custom_components/HA-FoxESS-Modbus/modbusLAN-H3.yaml

canton7, 2023-04-24:
Fixed in #167

closed 2023-04-24 by canton7