nathanmarlor/foxess_modbus #27: Recover after lost connection

FozzieUK, 2023-03-11:
Hi Nathan,

Just thought i'd post this on the off-chance there is something you can do to improve the recovery after failure.

It's basically a fault I created, I unplugged the RS485 connector from the W610 to check the error response when trying to write to a register.

After a short time I plugged the connector back in, and noticed after a few minutes that my sensors weren't updating - it looks like it hasn't been able to recover from that Fatal error - the only way back was to restart HA.

Is it possible to add some recovery code from a Fatal error where people do such stupid things which cause the connection to fail, or whether there may be another way of forcing the integration to re-open the connection without an HA restart.

Logs included below starting from just before I pulled the connector, the errors then continue after I reconnected the cable and took a restart to clear (many, many pages more of the logs if you need them but just repeats ad nauseam)

2023-03-10 16:16:30.280 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11048, 2)
2023-03-10 16:16:30.645 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11000:50)
2023-03-11 12:51:39.251 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11000, 8)
2023-03-11 12:51:39.551 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Refresh complete - notifying sensors
2023-03-11 12:51:39.631 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11008, 8)
2023-03-11 12:51:40.352 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11016, 8)
2023-03-11 12:51:41.072 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11024, 8)
2023-03-11 12:51:41.792 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11032, 8)
2023-03-11 12:51:42.512 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11040, 8)
2023-03-11 12:51:43.232 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11048, 2)
2023-03-11 12:51:43.942 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (41001:11)
2023-03-11 12:51:43.942 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (41001, 8)
2023-03-11 12:51:44.662 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (41009, 3)
2023-03-11 12:51:45.255 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11000:50)
2023-03-11 12:51:45.255 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11000, 8)
2023-03-11 12:51:45.378 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Refresh complete - notifying sensors
2023-03-11 12:51:45.452 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11008, 8)
2023-03-11 12:51:46.172 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11016, 8)
2023-03-11 12:51:46.262 ERROR (MainThread) [pymodbus.logging] Unable to decode response Modbus Error: Unknown response 0
2023-03-11 12:51:46.277 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/selector_events.py", line 876, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/base.py", line 342, in data_received
    self._data_received(data)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/base.py", line 395, in _data_received
    self.framer.processIncomingPacket(data, self._handle_response, unit=0)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 170, in processIncomingPacket
    self._process(callback)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 189, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2023-03-11 12:51:46.283 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Exception when polling modbus - Modbus Error: [Connection] Connection lost during request
2023-03-11 12:51:51.255 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11000:50)
2023-03-11 12:51:51.256 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11000, 8)
2023-03-11 12:51:51.462 ERROR (MainThread) [pymodbus.logging] Unable to decode response Modbus Error: Unknown response 0
2023-03-11 12:51:51.466 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/selector_events.py", line 876, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/base.py", line 342, in data_received
    self._data_received(data)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/base.py", line 395, in _data_received
    self.framer.processIncomingPacket(data, self._handle_response, unit=0)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 170, in processIncomingPacket
    self._process(callback)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 189, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2023-03-11 12:51:51.471 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Exception when polling modbus - Modbus Error: [Connection] Connection lost during request
2023-03-11 12:51:57.257 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11000:50)
2023-03-11 12:51:57.257 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11000, 8)
2023-03-11 12:51:57.942 ERROR (MainThread) [pymodbus.logging] Unable to decode response Modbus Error: Unknown response 0
2023-03-11 12:51:57.947 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/selector_events.py", line 876, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/base.py", line 342, in data_received
[... 51 lines cut ...]
    self._data_received(data)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/base.py", line 395, in _data_received
    self.framer.processIncomingPacket(data, self._handle_response, unit=0)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 170, in processIncomingPacket
    self._process(callback)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 189, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2023-03-11 12:52:15.472 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Exception when polling modbus - Modbus Error: [Connection] Connection lost during request
2023-03-11 12:52:21.261 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11000:50)
2023-03-11 12:52:21.261 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11000, 8)
2023-03-11 12:52:21.943 ERROR (MainThread) [pymodbus.logging] Unable to decode response Modbus Error: Unknown response 0
2023-03-11 12:52:21.947 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/asyncio/selector_events.py", line 876, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/base.py", line 342, in data_received
    self._data_received(data)
[... cut ...] ne 395, in _data_received
    self.framer.processIncomingPacket(data, self._handle_response, unit=0)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 170, in processIncomingPacket
    self._process(callback)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 189, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request

nathanmarlor, 2023-03-11:
Thanks Dave!

It's been coded so that every refresh is done independently, so if that one fails the next one has a chance of succeeding. The first thing every read/write request does is check if we're connected, and if not initiate a connection. With that, how long did you leave it to recover? I think eventually (~4-5 mins) pymodbus would realise the connection had dropped and close the socket, allowing a subsequent refresh to then reconnect and continue as normal.

FozzieUK, 2023-03-12:
Morning Nathan, 

I left it a couple of hours and it didn't recover - I'm using the W610 over LAN and i've noticed a few times that it isn't very good at releasing connections after an error (ends up with port 502 blocked), I should have tried modpoll to confirm it, i'll do that next time. 

I'm going to try out the Waveshare industrial LAN>RS485 converter this week the reviews suggest it's more robust than the W610 on LAN, hoping that's true ;)

nathanmarlor, 2023-03-12:
Thanks Dave - I think I know how I can get it to release the connection after a failure, I'll have a play this week.

FYI - latest beta now includes a write registers service if you fancy taking it for a test drive :) just a few issues to iron out with USB support before releasing proper. Also expecting the config flow to change again so you'll need to delete/re-add the integration over the next couple of releases.

FozzieUK, 2023-03-12:
Fantastic, i'll get that down loaded and test it now :)

FozzieUK, 2023-03-12:
Ok, did some testing on the write registers.

So far works fine on single register write (41010="75")
Works fine with multiple register write (41001="1,30,1054,1,1281,1538") charge times (00:30->4:30, 05:01->06:02)

I have an automation that I use on the old modbus write_registers service which uses a template to convert the value from a slider to update max_soc   

service: foxess_modbus.write_registers
data_template:
  start_address: 41010
  values: "{{ states('input_number.maximumsoc')|int(default=0) }}"

This doesn't work and it complains about expecting a string but received none, I tried changing data: to data_template:, I tried converting to a string but it still throws an error.
[image]

Any chance you could take a look and see if I can pass in a variable or result of a template please :)

nathanmarlor, 2023-03-12:
@FozzieUK this is working for me...

`show_name: true
show_icon: true
type: button
tap_action:
  action: call-service
  service: foxess_modbus.write_registers
  data:
    start_address: 41010
    values: '100'
  target: {}
entity: sensor.max_soc
show_state: true
`

FozzieUK, 2023-03-12:
@nathanmarlor That works when you have a fixed value '100' but if you replaced the value with a sensor value (or in my case a slider) it fails as it has to be evaluated by a template.

The old modbus write register service allowed me to have an automation that waited for a slider to move and then passed the value (evaluated by template) to the service.

This is fairly standard for Home Assistant, it's so much easier to code it than negotiate the vagaries of automations, scripts, templates etc..  ;)

But thanks, your example  got me thinking the custom button card allows manipulation in javascript so you can pre-process the value and insert it as a value before calling the service.

This works perfectly, basically move the slide then click the button to action it :)

```
type: custom:button-card
entity: sensor.max_soc
show_state: false
name: |
  [[[
        var bri = (states['input_number.maximumsoc'].state);
        var bri1 = (+bri).toString();
        return "slider set to " + bri1 + "%, Inverter maxSoC=" + (states['sensor.max_soc'].state) + "%";
  ]]]
styles:
  card:
    - height: 100px
    - width: 340px
![test](https://user-images.githubusercontent.com/113460294/224554245-771d8029-7a13-4731-aba0-1ccc0d6ffaa4.jpg)

tap_action:
  action: call-service
  service: foxess_modbus.write_registers
  service_data:
    start_address: 41010
    values: |
      [[[ 
        var bri = (states['input_number.maximumsoc'].state)
        var bri1 = (+bri).toString();
        return bri1 ;
      ]]]
```

[image]

nathanmarlor, 2023-03-12:
@FozzieUK ah, I see the issue! I'll have a play when I get 5 and see if I can get it working the old way again :)

nathanmarlor, 2023-03-13:
@FozzieUK re-connect after a fault is ready to roll in the latest beta - you should see it reconnect once you plug it back in. Still ironing out the last USB bug but should make a proper release by the end of the week.

FozzieUK, 2023-03-13:
@nathanmarlor Ok great i’ll give it a test drive tomorrow - i’ve used up all my solcast api calls today ;)

FozzieUK, 2023-03-14:
@nathanmarlor Morning Nathan, i've checked the disconnect/reconnect and all looks good, tried RS485 cable disconnect and W610 LAN disconnect both recover fine, i've put the logs in below but basically nothing to report, it works perfectly.

The modbus write service has inherited a **friendly_name key**, if you call it without setting it, it fails, I didn't know whether that was intentional or a side effect but easy to get over by specifying a blank key in the service data.
```
show_name: true
show_icon: true
type: button
tap_action:
  action: call-service
  service: foxess_modbus.write_registers
  data:
    start_address: 41010
    friendly_name: "" 
    values: '100'
  target: {}
entity: sensor.max_soc
show_state: true
```
I was going to mention i'm having the same problem with statistics and state_class not specified but looks like you're on to that :)

Recovery after RS485 cable pulled
>poll loop
2023-03-14 08:19:56.798 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11000:50)
2023-03-14 08:19:56.799 INFO (MainThread) [custom_components.foxess_modbus.modbus_client] Connecting to modbus
2023-03-14 08:19:56.803 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11000, 8)
2023-03-14 08:19:59.806 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Timed out when contacting device, cancelling poll loop
2023-03-14 08:20:02.801 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11000:50)
2023-03-14 08:20:02.802 INFO (MainThread) [custom_components.foxess_modbus.modbus_client] Connecting to modbus
2023-03-14 08:20:02.811 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11000, 8)
2023-03-14 08:20:02.974 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11008, 8)
2023-03-14 08:20:03.694 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11016, 8)
2023-03-14 08:20:04.414 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11024, 8)
2023-03-14 08:20:05.134 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11032, 8)
2023-03-14 08:20:05.854 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11040, 8)
2023-03-14 08:20:06.574 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11048, 2)

Recovery after LAN cable disconnected to W610
>[Connection] Error connecting to device)
2023-03-14 08:21:59.517 WARNING (MainThread) [pymodbus.logging] Failed to connect: [Errno 113] Connect call failed ('192.168.0.157', 502)
2023-03-14 08:21:59.521 WARNING (MainThread) [pymodbus.logging] Ignoring launch of delayed reconnection, another is in progress
2023-03-14 08:22:02.583 WARNING (MainThread) [pymodbus.logging] Failed to connect: [Errno 113] Connect call failed ('192.168.0.157', 502)
2023-03-14 08:22:02.825 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11000:50)
2023-03-14 08:22:02.825 INFO (MainThread) [custom_components.foxess_modbus.modbus_client] Connecting to modbus
2023-03-14 08:22:05.655 WARNING (MainThread) [pymodbus.logging] Failed to connect: [Errno 113] Connect call failed ('192.168.0.157', 502)
2023-03-14 08:22:05.657 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Connection] Error connecting to device)
2023-03-14 08:22:05.657 WARNING (MainThread) [pymodbus.logging] Failed to connect: [Errno 113] Connect call failed ('192.168.0.157', 502)
2023-03-14 08:22:05.660 WARNING (MainThread) [pymodbus.logging] Ignoring launch of delayed reconnection, another is in progress
2023-03-14 08:22:08.727 WARNING (MainThread) [pymodbus.logging] Failed to connect: [Errno 113] Connect call failed ('192.168.0.157', 502)
2023-03-14 08:22:08.826 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11000:50)
2023-03-14 08:22:08.826 INFO (MainThread) [custom_components.foxess_modbus.modbus_client] Connecting to modbus
2023-03-14 08:22:09.753 ERROR (MainThread) [pymodbus.logging] Factory protocol connect callback called while connected.
2023-03-14 08:22:09.762 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11000, 8)
2023-03-14 08:22:10.490 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11008, 8)
2023-03-14 08:22:11.210 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11016, 8)
2023-03-14 08:22:11.930 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11024, 8)
2023-03-14 08:22:12.650 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11032, 8)

nathanmarlor, 2023-03-14:
@FozzieUK perfect, glad it's working out! Friendly name is only there to cover multiple devices - similar to the native HA integration - as you've found 90% of us just need to leave this blank and it'll work as expected. I'll update the readme though! 

Closing this issue down for now, thanks as always for your input/feedback!

closed 2023-03-14 by FozzieUK

nathanmarlor, 2023-03-14:
@FozzieUK as a bonus, I've just fixed the service call issue so you can go back to your slider :)

FozzieUK, 2023-03-14:
@nathanmarlor Thanks, your a star! - i'll give that a go now :)

FozzieUK, 2023-03-15:
Hi Nathan,

Just reporting back the write_register service works perfectly with my slider now :)

Also just a quick point for discussion really to do with timing and poll rates on RS485 i'm guessing you've done a lot of this in testing already.

I was wondering how low I could make the aux polling to try and get it nearer the 5 seconds the old modbus used to manage on a few sensors (I know it managed it badly) - So I did the maths on the modbus message structures, all the messages together are less than 200mS and the inverter is fast to respond so I looked at the time my Pi4B takes to process the received registers and that is where the real delay comes in (300-500mS).

The way it is now configured with the 8 register per slice, means 9 requests to get all the data and each one has that heavy Pi processing time built in - the whole read register refresh process takes 5.8s for my Pi4.

If I reduce the poll time to 6s, there's no space left to make write_registers calls and it was taking nearly 30S for one to complete, at 8s it's ok - polling is fine, write_registers ok etc..

You originally had the number of registers read set at 10 per slice which gave those occasional misread errors (I assume that is a problem with the Pi processing as the modbus timings are easily up for it).

So was just wondered what you found in your testing, do you use a Pi4 or have you got a bit more bhp to hand? - it would be nice to get the polling time down to nearer the 6s if possible - and it should be possible if the registers reads were 10 per slice, but I think that is more a question of processing power to achieve it (roll on the Pi5!).

I'd go back to LAN direct but the extra registers are just too damn tempting ;)