nathanmarlor/foxess_modbus #30: Missing state_class

canton7, 2023-03-13:
## Version of the custom_component

v1.0.4

## Configuration

N/A

## Describe the bug

Firstly, thanks for the integration! 

I've noticed that some entities don't have a `state_class`, which means they can't be used on things like statistic graphs. The ones I've spotted so far are:

* pv1_power
* feed_in
* grid_consumption

(There are almost certainly others, but those are the ones I was trying to plot!).

Would it be possible to add a suitable `state_class` to all entities, please?

Thanks!

## Debug log

N/A

nathanmarlor, 2023-03-13:
Hi @canton7 - state classes are set -> https://github.com/nathanmarlor/foxess_modbus/blob/81eabbc31b1bdc86b890e995248c8995a4a43703/custom_components/foxess_modbus/sensors/h1_aux_sensors.py#L37

I can see statistics on my local dev environment - perhaps it takes a while to generate?

canton7, 2023-03-13:
Huh, that's odd.

I can see statistics from some entities provided by your integration (in Developer Tools -> Statistics), but they all have the error "The state class '' of this entity is not supported". pv2_power for example appears like this, but pv1_power doesn't appear in the statistics tab at all.

If I look at e.g. pv1_power under Developer Tools -> States, the state attributes shown are just:

 * unit_of_measurement
 * device_class
 * friendly_name

(whereas e.g. the equivalent entity provided by foxess-ha has `state_class: measurement` shown there).

Maybe it's a result of moving from HA-FoxESS-Modbus or something -- I'll try removing everything and re-adding tonight.

nathanmarlor, 2023-03-13:
You're absolutely right. I just looked with fresh eyes and noticed that despite being set in the description, it isn't set in the sensor 🤦 I'll resolve in the next release, thanks for the report!

canton7, 2023-03-13:
Aah brilliant, thanks for taking another look!

nathanmarlor, 2023-03-14:
@canton7 resolved in #36

closed 2023-03-14 by nathanmarlor