nathanmarlor/foxess_modbus #61: Are Ambient Temp and Inverter Temp correct registers

Stephens-ely, 2023-03-21:
When comparing the registers used for this FoxESS modbus integration I've noticed that the registers used for Ambient Temp and Inverter Temp are revered when compared to the integration offered by StealthChesnut.

I'm not in a position to say which is correct so am just pointing out the difference;

In https://github.com/StealthChesnut/HA-FoxESS-Modbus/wiki/Data-Register-Reference---H1-AC1

11024 = InvTemperation
11025 = AmbientTemperature

In this integration as defined in xx1_aux_sensors.py

    SensorDescription(
        key="invtemp",
        address=11025,
        name="Inverter Temp",
        device_class=SensorDeviceClass.TEMPERATURE,
        state_class=SensorStateClass.MEASUREMENT,
        native_unit_of_measurement="°C",
        scale=0.1,
    ),
    SensorDescription(
        key="ambtemp",
        address=11024,
        name="Ambient Temp",
        device_class=SensorDeviceClass.TEMPERATURE,
        state_class=SensorStateClass.MEASUREMENT,
        native_unit_of_measurement="°C",
        scale=0.1,
    ),

These are for the USR/USB connection

nathanmarlor, 2023-03-22:
Good spot, checked against Fox Cloud and fixed in #68

closed 2023-03-22 by nathanmarlor

Ndrinta, 2023-03-22:
If so, please fix it in `xx1_lan_sensors.py` if necessary.
Thanks

nathanmarlor, 2023-03-22:
@Ndrinta can you confirm the values are switched vs. fox cloud on LAN please? My LAN isn't plugged in anymore.

Ndrinta, 2023-03-22:
They are indeed.

[image]
[image]