nathanmarlor/foxess_modbus #66: Investigate larger reads with the W610

canton7, 2023-03-22:
I'm [playing around with modpoll](https://github.com/StealthChesnut/HA-FoxESS-Modbus/wiki/USR-W610-Setup-Guide#troubleshooting-usr) and the W610, and I can fairly reliably perform large (50 register) reads, provided that I pick the right starting register. 

E.g. I can poll 76 registers starting at 11020, but only a 75-register read from 11021. I can't read 11096 at all, even with a 1-register read.

So, I'm beginning to suspect that the issue is that some registers just aren't readable. With a larger read window, we were happening to hit registers which aren't readable.

This needs some more testing, but I think we could probably hugely increase the read size if we're careful to avoid unreadable registers.

nathanmarlor, 2023-03-22:
Interesting - on the W610 I can't read > 10 reliably.

[image]

[image]

canton7, 2023-03-24:
Turns out this is specific to the Windows version of modpoll. When the W610 returns an incomplete frame, as it does if you try and read more than 8 registers, the Windows modpoll just shows garbage. The Linux version at least shows 0's (and pymodpoll throws an error).

So, I was being fooled by what is presumably uninitialized memory somewhere... Oh well.

closed 2023-03-24 by canton7