nathanmarlor/foxess_modbus #70: Added sensor

manuele84, 2023-03-22:
Congratulations for the project, is it possible to add the following sensor?

Ind: 11011		aux inverter

and a sensor that indicates the alarm status of the inverter.

nathanmarlor, 2023-03-22:
@manuele84 can you add a little more description of what this sensor is/does please? :)

nathanmarlor, 2023-03-24:
@manuele84 :)

manuele84, 2023-03-24:
> @manuele84 :)
Sorry for the delay, the sensor calculates the power that comes out of the inverter in kw.

```
    - name: "Exit inverter"
      scan_interval: 5
      slave: 247
      address: 11011
      state_class: measurement
      unit_of_measurement: "kW"
      data_type: int16
      scale: 0.001
      precision: 3
      input_type: input
      device_class: power

```

nathanmarlor, 2023-03-24:
@manuele84 thank you, will add it in asap 😊

canton7, 2023-03-24:
(or rename the yield ones to this)

canton7, 2023-03-25:
Looking more, the numbers are identical to Feed-In. Which makes sense: the foxess app calls this RPower, and that makes sense as the register is right next to RVolt.

So we've just found another way of fetching Feed-in, which we were getting from the Grid CT anywya.

canton7, 2023-03-25:
Although, coming back to it this afternoon, and the values of 11011 aren't matching feed-in any more.

manuele84, 2023-03-25:
[image]

11011 is the Power exit inverter

nathanmarlor, 2023-03-25:
Joining the pieces up - I'm guessing the 'inverter power' (or whatever we call it) matches feed in if you don't have batteries and PV isn't yet generating as it's AC -> AC. The difference between this and feed-in will be inverter efficiency losses due to the DC -> AC conversion. Makes sense?

nathanmarlor, 2023-03-29:
Thanks @manuele84 - you put me on the right path to rationalising all the battery/inverter sensors. Now added in #101

closed 2023-03-29 by nathanmarlor