nathanmarlor/foxess_modbus #74: Missing CT2 sensor H1 LAN

FozzieUK, 2023-03-23:
The H1 LAN sensors are missing the CT2 meter sensor, it doesn’t appear to have been carried over from the original modbus, be grateful if you could add this back in.

    SensorDescription(
        key="ct2",
        address=31015,
        name="CT2",
        device_class=SensorDeviceClass.POWER,
        state_class=SensorStateClass.MEASUREMENT,
        native_unit_of_measurement="kW",
        scale=0.001,
    ),

canton7, 2023-03-23:
It's there, for H1 and AC1? https://github.com/nathanmarlor/foxess_modbus/blob/main/custom_components/foxess_modbus/entities/xx1_aux_sensors.py#L234-L242

canton7, 2023-03-23:
Looks like neither CT1 or CT2 are there. They're missing [from here](https://github.com/StealthChesnut/HA-FoxESS-Modbus/wiki/Data-Register-Reference---H1-AC1), although they are [here](https://github.com/StealthChesnut/HA-FoxESS-Modbus/blob/09936e70e0e2c56a8c5730a70e431c47edefb6d5/custom_components/HA-FoxESS-Modbus/modbusLAN.yaml#L149-L160)

FozzieUK, 2023-03-23:
I think it was added to the stealthchestnut modbus earlier this year but looks like the wiki hasn’t been kept up to date.

canton7, 2023-03-23:
I've done a pass through modbusLAN.yaml, and it looks like we've caught everything else.

FozzieUK, 2023-03-23:
Thats great, thanks

PR #76 (2023-03-24): Add CT2 to H1/AC1 LAN
Fixes: #74
commit cdf3640a (2023-03-24): Add CT2 to H1/AC1 LAN

Fixes: #74
closed 2023-03-24 by nathanmarlor