nathanmarlor/foxess_modbus #78: Entity not defined

andykorol1, 2023-03-24:
Hi after updating to the this release of modbus none of the entity's appear to work with the energy dashboard? I've attached an example of the error it gives me.

[image]

canton7, 2023-03-24:
None of this integration's sensors start with `foxess` (although I rather suspect we should have a prefix, but that's a different matter...)

`foxess_bat_charge` is provided by the foxess-ha integration, at least on my setup.

andykorol1, 2023-03-24:
I've been trying every sensor I could think of. Just tried to copy and paste the sensor name from this modbus integration and it doesn't show up at all.

nathanmarlor, 2023-03-24:
It'll only accept a sensor with Total/Total_Increasing. You could either setup a Reimann sum integration, or wait for the latest release :)

[image]

canton7, 2023-03-24:
It looks like the dashboard wants sensors which provide a total value in kWh, and this integration currently just provides the current power in kW.

That said, #79 has literally just this minute added support for the sensors you need, so wait for the next release and you should be good. Alternatively you can add your own riemann integral helper which integrates the power to get kWh.

nathanmarlor, 2023-03-24:
[image]

Closing in advance of the latest release.

canton7, 2023-03-24:
This is out as the pre-release version [v1.4.0b3](https://github.com/nathanmarlor/foxess_modbus/releases/tag/v1.4.0b3), which you should be able to install by checking "show beta versions" in HACS