nathanmarlor/foxess_modbus #85: Daylight savings time-related weirdness?

canton7, 2023-03-26:
## Version of the custom_component

v1.4.0b3

## Describe the bug

So, this morning I had a fairly large spike of feed-in (purple), without any corresponding generation (blue), which seemed a little odd. 

[image]

So I went looking into the logs. The only actual error around that time (I don't have debug logs enabled) is:

```
2023-03-26 06:34:00.495 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 543, in async_update_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 597, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 562, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 855, in state
    value = self.native_value
  File "/config/custom_components/foxess_modbus/entities/modbus_charge_period_sensors.py", line 74, in native_value
    value = time(hour=(value & 0xFF00) >> 8, minute=value & 0xFF)
ValueError: minute must be in 0..59
2023-03-26 06:35:10.243 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.grid_consumption_sum_today, value dropped from 1.1 to 2023-03-26T05:34:00.494176+00:00, triggered by state with last_updated set to 0.0
```

The first oddity there is of course that we managesd to get a number of minutes > 0 on the start/end of a charge period (which should be set to midnight I should add, since they're both disabled).

The second is that HA seems to think that grid_consumption_today fell from 1.1 to... a date? If I go looking, it fell to 0.0 for about an hour: 

[image]

It looks like charge period 2 was briefly enabled:

[image]

With a start time of 00:15 and an end time of midnight:

[image]

Min SoC took a dive to -100%:

[image]

I'm not sure what to make of all of that, whether it's the inverter messing up, or HA getting confused. I can't see how grid_consumption_today could have become a date...

It's also an odd time, since the clocks go forward around 3am, not half 6.

canton7, 2023-03-26:
Interestingly, foxesscloud looks like it failed as well (that looks an awful lot like interpolation across the weird region):

[image]

So my current best guess is that the inverter did a time sync at 6:34, found out the clocks had gone forwards, and crapped itself for about half an hour. I'm not sure what happened to those poor time sensors though...

nathanmarlor, 2023-03-26:
Very odd indeed - unfortunately/fortunately (😁) I didn't see any weirdness last night. I can't think of a reasonable reason as to why the time updating would cause the level of data corruption you've experienced though...?! And how/why it recovered after an arbitrary period of time. Might be worth adding inverter time sensors in preparation for next year 🤪 - I think they're at 40000-40005. 

[image]

canton7, 2023-03-26:
Yeah, I'm not sure there's much we can do.

I do want to add better validation to a bunch of stuff - log more detailed info if we read percents outside of 0-100, etc, then hopefully if it happens again we can get some better info. I've done that for the charge period start/end, but I'll add others as well.

canton7, 2023-03-26:
I also wonder whether the W610 read a bunch of registers at an offset of something. I can do some digging and see if there's a register near SoC which might have that value...

canton7, 2023-03-26:
Actually, I do see my soc at - 2% quite often, despite not having a battery yet. I assumed it was just confused, but maybe that's the tip of something bigger, and related

Ndrinta, 2023-03-26:
I'm running 1.4.0b3 too and I woke up 2 mornings on a row with 0W/0A for the solar panels panels related sensors, while everything else seemed working fine.
A reload of the integration fixed the issue and yesterday I did not want to raise any flags because it's a beta version and it could have been a glitch, but 2 I'm a row... Not sure if it's related to this but I believe it's relevant.

canton7, 2023-03-27:
@Ndrinta Can you please check your logs to see whether there's anything from this integration, and turn on debug logging to catch more info if it happens again?

nathanmarlor, 2023-03-29:
@Ndrinta before the logs overwrite please :smile:

Ndrinta, 2023-03-29:
Sorry, I forgot to report on this.
I could't find absolutely nothing on the actual log (which goes back to the 24th and I see that happening at least 3 times).
I did not turn on the debug logging for the integration 2 days ago.
I've just done it now and wait until happens again, meanwhile I can confirm i see foxess_modbus's verbose on the Log.
I've also took the opportunity to update to b5.

canton7, 2023-03-30:
I just noticed that I'm still seeing odd values some early mornings (including invalid times which the latest validation isn't catching - I'll have to take a look at those)

canton7, 2023-04-06:
I'm still getting odd values. I think they're related to the inverter getting stuck in a loop of powering off and on at the start of dawn / end of dusk, when there isn't quite enough power from the panels to power it, and I *think* that's interacting with #122 which means that we end up with bad values being presented for an extended period... I really need to either get some better logging in or catch it in the act though!

nathanmarlor, 2023-04-06:
Do you have idle mode set?

Ndrinta, 2023-04-06:
> Actually, I do see my soc at - 2% quite often, despite not having a battery yet. I assumed it was just confused, but maybe that's the tip of something bigger, and related

I did never see values outside of the ranges in my instance. Actually, it has always either been perfectly fine or simply 'Unavailable'.
I think you are looking at something related to your local changes... Did you have any at the time?

canton7, 2023-04-06:
> Do you have idle mode set?

Idle mode? Bear in mind I don't have a battery yet, which is known to cause the boot loop in minimal light conditions.

Ndrinta, 2023-04-06:
It has behaved itself for a while now, this after reverting the poll rate back to 10s (I ventured exploring faster polling rates with the W610, but was failing in the long term).
I've just swapped to the new waveshare and immediately crank it down to 3s, so far so good but I'll see if I experience the same tomorrow morning.

nathanmarlor, 2023-04-06:
> > Do you have idle mode set?
> 
> Idle mode? Bear in mind I don't have a battery yet, which is known to cause the boot loop in minimal light conditions.

[image]

Allows the inverter to 'shut down' when not generating PV/battery energy.

canton7, 2023-04-22:
I haven't seen any more weirdness since #122 went in, so I'm closing this off

closed 2023-04-22 by canton7