nathanmarlor/foxess_modbus #87: Install notes for readme

FozzieUK, 2023-03-26:
It may be helpful to add a few notes to the **Readme** for people moving to this integration from the StealthChestnut version as there are a number of similar problems occurring. Hopefully adding some notes to do things in a certain order, and remove certain sensors that can be duplicated would help to avoid this.
I suspect this might be a moving target for a while but starting with the most common ones.

>**NOTE:** When transferring away from the StealthChestnut HA-FoxESS-Modbus version, **before** installing this integration please remove the modbus: include from the configuration.yaml, it will look similar to this (depending on the path you chose)
>**modbus: !include custom_components/foxmodbuslan/modbusUSB.yaml** 
>
>**AND** if you have added the sensors listed below, remove them from your configuration.yaml as they are now provided from within the integration
```
  - name: "Battery Discharge"
    device_class: "power"
    unit_of_measurement: "kW"
    state: >
       {% if (states('sensor.battery_discharge_power') | float(default=0) ) > 0 %}
       {{ states('sensor.battery_discharge_power') | float(default=0) * 1 }}
       {% else %}
       0 
       {% endif %}
  - name: "Battery Charge"
    device_class: "power"
    unit_of_measurement: "kW"
    state: >
       {% if (states('sensor.battery_discharge_power') | float(default=0) ) < 0 %}
       {{ states('sensor.battery_discharge_power') | float(default=0) * -1 }}
       {% else %}
       0 
       {% endif %}
  - name: "Grid Consumption"
    device_class: "power"
    unit_of_measurement: "kW"
     state: >
       {% if (states('sensor.grid_ct') | float(default=0) ) < 0 %}
       {{ states('sensor.grid_ct') | float(default=0) * -1 }}
       {% else %}
       0 
       {% endif %}
```
>Save the configuration.yaml, Check Configuration is OK and Restart Home Assistant
>
>Now install the FoxESS modbus integration via HACS

canton7, 2023-03-26:
To be honest, it might be worth adding a step tithe config flow. "Migrating from....? Read these docs first" with a link.

FozzieUK, 2023-03-26:
Yep seems a shame to divert attention from the actual core install process, these are ‘before you install steps’ - so maybe a wiki or something similar

canton7, 2023-03-26:
Yeah, but it's our last chance to stop someone before they create duplicate sensors by accident.

Alternatively, we can rename ours so the don't conflict, but that does make migrating a bit harder...

nathanmarlor, 2023-03-26:
We could add a background step to check a cross section of existing sensors to ensure they don't still exist and alert if they do? Therefore if the migration steps have been followed the user sees nothing, if not we can end the config flow and redirect to the wiki/readme? 

I'll have some time mid-week so will look at re-jigging the config flow as per our discussions @canton7 (TCP/UDP/LAN etc) and see if I can squeeze something like this in.

canton7, 2023-03-26:
Yep, that sounds like a better plan!

StealthChesnut, 2023-04-05:
After switching over from [HA-FoxESS-Modbus](https://github.com/StealthChesnut/HA-FoxESS-Modbus), I found it reasonably straightforward. I do have a few extra sensors, but for the most part this worked as a high-level approach:
* Disable the yaml config line for modbus include
* Disable any related sensors
* Install this integration and run through setup
* re-enable missing sensors as needed

@FozzieUK 's advice works well, I second having that as a line / collapsed section in the Readme would be useful.

nathanmarlor, 2023-04-17:
@FozzieUK done, does that look OK?

FozzieUK, 2023-04-17:
@nathanmarlor looks good to me thanks 👍☺️

closed 2023-04-17 by FozzieUK