nathanmarlor/foxess_modbus #93: Setup Workflow - Error while setting up Serial option resets the configuration form to TCP option

StealthChesnut, 2023-03-28:
While setting up the integration I chose Serial as my option, got one of my settings wrong and it couldn't connect to my USB device (this is fine and expected).

The form then reset, but it is now the TCP setup form (asking for a Host and Port).

I can close it and start over, but it would be nice if it knew to reset to the option chosen by the user when starting the setup workflow.

nathanmarlor, 2023-03-28:
I think this was fixed in #67 - which version are you on?

StealthChesnut, 2023-03-28:
I've installed onto a fresh HA install today, so the latest/current version as presented - v1.3.1

nathanmarlor, 2023-03-28:
@canton7 @StealthChesnut can you sense check #96 please?

canton7, 2023-03-28:
I don't know that code well enough to give it a proper review, but I gave it a spin and couldn't break it.

nathanmarlor, 2023-03-28:
I don't know why I'd made it so complex before 🤷