nathanmarlor/foxess_modbus #95: Problems with write_registers service

canton7, 2023-03-28:
## Version of the custom_component

v1.4.0b4

## Describe the bug

Just having a play with the write_registers service.

1. It seems that the friendly_name is required, even if the user didn't enter a friendly_name during configuration. Is that intentional? At least, the UI tells me that the field must be filled in
2. I get the following error in the log when trying to write 41001

```
2023-03-28 13:56:24.507 WARNING (MainThread) [custom_components.foxess_modbus.services] Modbus Error: [Input/Output] Error writing holding register: Exception Response(134, 6, IllegalAddress)
Traceback (most recent call last):
  File "/config/custom_components/foxess_modbus/services/write_registers_service.py", line 47, in _write_service
    await controller.write(service_data)
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 66, in write
    await self._write_registers(start_address, values)
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 74, in _write_registers
    await self._client.write_registers(start_address, values, self._slave)
  File "/config/custom_components/foxess_modbus/modbus_client.py", line 107, in write_registers
    raise ModbusIOException(f"Error writing holding register: {response}")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Error writing holding register: Exception Response(134, 6, IllegalAddress)
```

nathanmarlor, 2023-03-28:
The first is a bug I've been meaning to raise.

The second is due to the fact that 41001 -> 41006 need to be written together as a group of 6.

canton7, 2023-03-28:
Aha, *that's* the error it causes. Thanks!

closed 2023-03-28 by nathanmarlor