nathanmarlor/foxess_modbus #106: Friendly name setting

nathanmarlor, 2023-04-02:
Friendly names aren't escaped (i.e. ' ' -> _) so result in erroneous sensors.

canton7, 2023-04-02:
Would it be worth creating both a friendly name and friendly id in config? That let's us e.g. change how it is escaped over time, without breaking people.

(It also lets me have a prefix to all the sensor ids, to keep things neat, without having a friendly name appear at the end of every sensor name)

iancg, 2023-04-02:
You need the friendly name in the sensor description if you have two inverters otherwise the graphs just show the same sensor names twice and you can't tell which is which (it was me that asked for that to be added).

canton7, 2023-04-02:
Right, but I've only got 1 inverter. The aim, which I think other people are doing as well, is to put a prefix onto all of the entity ids (which we don't have my default, to be compatible with StealthChestnut's intigration) - this makes it easier to e.g. use globs to slurp everything into influxdb, rather than having to list every sensor individually.

That isn't what this ossie is about, of course. But I saw the opportunity to kill two birds with one stone here.

canton7, 2023-04-02:
We also need to make sure that two friendly names don't escape to the same friendly id. That's probably also easier to sort out if we generate both at config time, rather than having to sort it out on the fly each time.

iancg, 2023-04-03:
It's a good point.  Without wanting to over-complicate the config, I wonder if we need a separate "entity_id prefix" from "Friendly name prefix or suffix"?

With StealthChestnut's integration I'd actually prefixed my Friendly name with the inverter name and the suffix was @nathanmarlor 's alternative implementation (but since the entity id's were prefixed, it mattered not).  I'd mainly used prefix as some of the entity_id's are generated from the escaped name in StealthChestnut's one in some cases - e.g. 

   - name: "Inner-PV Energy Daily"

and I didn't want a mix of prefix and suffix in the entity_id's.

screenagerbe, 2023-04-04:
It's something I'm struggling with too...

Got StealthChestnut's integration up and running in my PROD environment, but I kind of modified it a lot. I've renamed all name's (added `FoxESS` as prefix) and unique_id's (added `foxess_` as prefix) entries in configuration.yaml, modbusLAN.yaml and all the extra entities which are listed in sensors.yaml and templates.yaml. 
Main reason: The FoxESS installation is not the only monitoring tool in my fuse box and I wanted them as recognizable as possible on all fronts.
And in the near future there will also be data send from a P1-port to HA.

I now got a WaveShare RS485 PoE to ETH (b) installed and configured in my TST environment and I'm wondering how I'm going the handle the naming thing 🤔🤨

canton7, 2023-04-04:
That should be the simple case of just setting `foxess` as your friendly name? Then it'll get added as a prefix to all your entity ids, and as a sufix to all entity names.

screenagerbe, 2023-04-04:
Yes it does the trick, thanks. (Wasn't really sure if that naming thing did what you said it would)
I just need to check if I don't get those silly _2 sensors, but probably that won't be avoidable

canton7, 2023-04-09:
As a short term fix for the next release, we could just add validation to only allow `[a-z_]`

canton7, 2023-04-09:
Ah, looks like homeassistant.core had a valid_entity_id function

canton7, 2023-04-15:
Right, I've put a limit on friendly names being snake_case in #139. That at least stops people unwittingly using values which won't work.

I've reworked this bit of the config flow in #139. I haven't added any more settings here, but it's worth waiting until #139 is in before trying to change anything else.

closed 2023-04-17 by nathanmarlor