nathanmarlor/foxess_modbus #110: BMS Watthours sensor

FozzieUK, 2023-04-03:
Just out of interest i've been getting to the bottom of 'what' the watthours sensor actually is.

I created a watthrate sensor to track it and confirm its value, I had to remove the scale to bring it in line.

[image]

The only thing I wasn't sure of, was whether the counter was 16 or a 32 bit register - so tracked it for the last few, it eventually got to 65535 and rolled to 0 without any adjacent register incrementing, confirming it is 16 bit, and simply a counter used internally by the BMS as an energy meter to track all battery activity in watts.

So the only changes are the scale 0.1 is incorrect, and should be removed, and the unit of measurement is Wh (not kWh)

https://github.com/nathanmarlor/foxess_modbus/blob/137329b9c8056e0fb73eeaafe5cecfa2cef70abd/custom_components/foxess_modbus/entities/xx1_aux_entity_descriptions.py#L588-L589

canton7, 2023-04-03:
Nice work! So it's the integral of all (absolute) battery charge and discharge? 

It's a shame that it wraps at 16 bits -- I wonder whether it's worth trying to unwrap that in HA...

Maybe we should /1000 and keep the unit as kWh, for consistency with other energy sensors?)

FozzieUK, 2023-04-03:
@canton7 yes that's it exactly 👍 

I was really hoping it was going to be an int32 :cry:  - but it's connected to the battery cycle counter, the BMS must do some internal maths on the capacity of batteries installed and increment the cycles so it's probably not worth re-creating a bigger number in HA as you could get there (roughly) by taking your (cycles * known capacity) + watthours.

Actually /1000 would make a lot of sense to keep in kWh (and help the HA database) :smile:

canton7, 2023-04-05:
To be honest, do we even want this sensor? Is it of any use to anyone?

canton7, 2023-04-22:
This is now in

closed 2023-04-22 by canton7