nathanmarlor/foxess_modbus #115: More validation issures

FozzieUK, 2023-04-04:
I just noticed my pv2 power sensor kept going unavailable - I thought it might have been comms but the rest of the sensors were fine - checked the logs and found this

[image]

It’s the good old slightly negative readings you sometimes get on the pv1/2 power strings but the validation (range 0) stops it working and makes it unavailable.

I’ve made a quick change in my code for the validation ranges on pv1 & 2 power (reg’s 11002 & 11005) to start range at -1 and it’s back now showing -0.001kwh
[image]

I’m pretty sure these are the only PV sensors that ever go negative, can we change the validation ranges please on 11002 & 11005 to start at -1 please :smiley:

canton7, 2023-04-04:
I get the same for battery voltage, without a battery attached. I say we remove validation on these - we're pretty sure we're interpreting the values correctly, and all we're doing is showing unavailable instead of what the inverter reports.

FozzieUK, 2023-04-04:
@canton7 that works for me 👍

closed 2023-04-06 by nathanmarlor