nathanmarlor/foxess_modbus #121: Add missing registers

canton7, 2023-04-05:
<body link="#0563C1" vlink="#954F72">

31010 | 11015 | EPS RVolt | 2376 | 237.6 V | int16 | 0.1 | V | voltage
-- | -- | -- | -- | -- | -- | -- | -- | --
31011 | 11016 | EPS RCurrent | 2 | 0.2 A | int16 | 0.1 | A | current
31012 | 11017 | EPS RPower | 3 | 0.003 kW | int16 | 0.001 | kW | power

</body>

From https://github.com/StealthChesnut/HA-FoxESS-Modbus/issues/112#issuecomment-1497841670

FozzieUK, 2023-04-06:
If you don't have EPS it's more registers to add, more to poll (thinking the slow W610) and more to exclude from your database to avoid bloat.

I would rather see an option in the config for 'Add EPS sensors?' than add the sensors, but it's a lot more work, more complexity and tbh I would rather have them than not so feel free to ignore this comment.

canton7, 2023-04-06:
Yeah... I would eventually like to get to a point where the user could disable a sensor, and we would stop polling for it. Then we could have sensors which are disabled by default, and the user could enable them if they wanted.

canton7, 2023-04-12:
I've also found:

10016: master version
10017: slave version
10018: manager version

canton7, 2023-04-24:
EPS implemented in #167, versions covered by #176

closed 2023-04-24 by canton7