nathanmarlor/foxess_modbus #127: Cumulative sensors not showing any data

andrewmooreio, 2023-04-06:
## Version of the custom_component

[v1.4.0b7](https://github.com/nathanmarlor/foxess_modbus/releases/tag/v1.4.0b7)

## Configuration

```yaml
No logs.
```

## Describe the bug

I've just switched over from the StealthChesnut integration to this. The following sensors are showing as unknown and have no data:

- Load Power Total (sensor.load_power_total)
- PV1 Power Total (sensor.pv1_energy_total)
- PV2 Power Total (sensor.pv2_energy_total)

I appreciate I'm on a beta version so can't expect everything to be perfect. Thank you!

## Debug log

```text

No logs.

```

canton7, 2023-04-06:
Is there nothing at all in your logs?

Do you have a H1 or an AC1?

canton7, 2023-04-06:
Ah, do you have a friendly name set which has a space in it? If so, this is #106.

nathanmarlor, 2023-04-06:
Which model of inverter are you using please?

andrewmooreio, 2023-04-06:
My inverter is a FoxESS H1-5.0-E. There are no errors in the logs relating to the FoxESS - Modbus integration.

I've enabled debug logging and I just see logs of entries for each register being read, but nothing untoward.

The friendly name does indeed have spaces in it.

canton7, 2023-04-06:
Yep, #106. That's due to be fixed before the next proper release.

andrewmooreio, 2023-04-06:
Thanks! I'll change the friendly name and hopefully that fixes it.

canton7, 2023-04-06:
Yep, it just needs to not have a space in it. HA silently fixes things for us most of the time, but not for the source entity of an integration sensor.

nathanmarlor, 2023-04-06:
I'd expect all sensors to blow up?

canton7, 2023-04-06:
I think HA must escape it for us most of the time.

andrewmooreio, 2023-04-06:
The 3 sensors mentioned in the OP are working now I've omitted the friendly name. Thanks both.

closed 2023-04-06 by nathanmarlor

canton7, 2023-04-06:
Duplicate of #106