nathanmarlor/foxess_modbus #139: Improve config flow

canton7, 2023-04-10:
Following on from #40.

The idea is to let someone select the adapter they're using during the config flow, and then set read sizes and poll rates based on that. We also let the user select between TCP and UDP if relevant, and provide links to adapter-specific documentations.

canton7, 2023-04-10:
I'm working on this, but it might take a little while. So far:

Entry page:
[image]
Direct LAN connection:
[image]
W610:
[image]
Waveshare:
[image]

iancg, 2023-04-10:
Is the entry page the page after you choose USB or IP connection? If not, how do you select a USB adapter?

canton7, 2023-04-10:
I haven't put adapters in for that yet, but they'll be in the same list. Along with "Other USB" and "Other Ethernet" or similar

iancg, 2023-04-10:
Ok - I've got DSD TECH SH-U10 USB (https://amzn.eu/d/1MDqSip), but I've also seen quite a few other USB and IP adaptors mentioned in the forums (often the AliExpress cheap adapters seen to work as well as the more expensive branded adapters)

canton7, 2023-04-10:
Yep, the list isn't complete. And quire honestly there's only really a point in putting adapters where we know the max read size, otherwise we'll just have to apply a blanket slow but safe config

iancg, 2023-04-10:
Yeah, I think all the USB adapter's will be similar as the baud rate of the inverter is limited to 9,600 so we only get about 960 characters a second at half duplex.

canton7, 2023-04-12:
@iancg Would you mind seeing what size reads you can do with that adapter? https://github.com/nathanmarlor/foxess_modbus/discussions/40#discussioncomment-5591666

iancg, 2023-04-12:
Starting point (default 10s/8) with two inverters, I get consistent errors, but suspect it is not related to read size:

    grep "custom_components.foxess_modbus" home-assistant.log | grep -B 1 "Modbus Error"
    2023-04-12 15:51:02.965 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11008, 8, (247))
    2023-04-12 15:51:08.989 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
    --
    2023-04-12 15:51:12.963 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11008, 8, (247))
    2023-04-12 15:51:18.987 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
    --
    2023-04-12 15:51:22.966 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11008, 8, (247))
    2023-04-12 15:51:28.986 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
    --
    2023-04-12 15:51:33.069 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11008, 8, (246))
    2023-04-12 15:51:39.096 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
    --
    2023-04-12 15:51:42.972 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11008, 8, (247))
    2023-04-12 15:51:48.991 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
    --
    2023-04-12 15:51:52.981 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11008, 8, (247))
    2023-04-12 15:51:58.999 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response

iancg, 2023-04-12:
Upping to 16 appears to work, but with similar errors, but now on a different register (do some registers need to be read in groups in the same way some need to be written in groups?)

    2023-04-12 15:57:14.252 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11000, 16)
    2023-04-12 15:57:14.252 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11000, 16, (247))
    2023-04-12 15:57:14.253 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11000, 16)
    2023-04-12 15:57:14.253 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11000, 16, (246))
    2023-04-12 15:57:14.349 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11021, 16)
    2023-04-12 15:57:14.349 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11021, 16, (247))
    2023-04-12 15:57:20.367 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
    2023-04-12 15:57:20.474 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11037, 13)
    2023-04-12 15:57:20.474 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11037, 13, (247))
    2023-04-12 15:57:20.571 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11069, 16)
    2023-04-12 15:57:20.571 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11069, 16, (247))
    2023-04-12 15:57:20.682 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11085, 2)
    2023-04-12 15:57:20.683 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11085, 2, (247))
    2023-04-12 15:57:21.435 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (41000, 12)
    2023-04-12 15:57:21.435 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (41000, 12, (247))
    2023-04-12 15:57:21.526 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Refresh complete - notifying sensors: {11009, 11011, 11045, 11014, 11021, 11023, 11024, 11000, 11002, 11003, 11005}

canton7, 2023-04-12:
Interesting... That "No response received, expected at least 4 bytes (0 received)" is similar to the W610 when it fails to return enough results. 

Odd that it work for larger reads. I think there might be some registers we have to avoid, but I'm pretty sure there are none in that range.

Does it work consistently with small reads, like 1 or 2?

canton7, 2023-04-12:
It's also odd that with a read size of 8, it still manages to read the first set of 8 fine then fails on the second.

Presumable the UI is showing lots of unknowns / stale data?

iancg, 2023-04-12:
Seems to be very consistent though - same registers for both inverters
  
    2023-04-12 16:06:14.458 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11034, 16, (246))
    2023-04-12 16:06:20.477 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
    --
    2023-04-12 16:06:24.327 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11034, 16, (247))
    2023-04-12 16:06:30.342 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)

But fails on one inverter much more often than the other (assuming both are polled at the same frequency)

    [core-ssh config]$ grep "custom_components.foxess_modbus" home-assistant.log | grep -B 1 "Modbus Error" | grep -c "(11034, 16, (247))"
    46
    [core-ssh config]$ grep "custom_components.foxess_modbus" home-assistant.log | grep -B 1 "Modbus Error" | grep -c "(11034, 16, (246))"
    7

iancg, 2023-04-12:
One register, still errors, but different register again:

    --
    2023-04-12 16:11:21.022 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11001, 1, (247))
    2023-04-12 16:11:27.032 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
    --
    2023-04-12 16:11:31.330 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11002, 1, (247))
    2023-04-12 16:11:37.342 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
    --
    2023-04-12 16:11:41.037 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11001, 1, (247))
    2023-04-12 16:11:47.052 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
    --
    2023-04-12 16:11:51.082 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11001, 1, (247))
    2023-04-12 16:11:57.096 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
    --
    2023-04-12 16:12:01.096 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11001, 1, (247))
    2023-04-12 16:12:07.116 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
    --
    2023-04-12 16:12:11.037 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11001, 1, (247))
    2023-04-12 16:12:17.054 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
    --
    2023-04-12 16:12:21.132 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11001, 1, (246))
    2023-04-12 16:12:27.147 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)

canton7, 2023-04-12:
Curious, that time it's failing consistently with the same slave. In previous times, it was switching between them

canton7, 2023-04-12:
Are you able to get data into the UI at all? I see a few "notifying sensors" messages in the 32-register reads, but not in the others

iancg, 2023-04-12:
Think I just hadn't waited long enough for a full set of results - but it doesn't fail every time by the look.

    2023-04-12 16:15:31.804 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11001, 1, (246))
    2023-04-12 16:15:37.821 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
    --
    2023-04-12 16:15:41.018 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11000, 1, (246))
    2023-04-12 16:15:44.435 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
    --
    2023-04-12 16:15:51.018 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11000, 1, (246))
    2023-04-12 16:15:51.113 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
    --
    2023-04-12 16:15:51.845 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11001, 1, (247))
    2023-04-12 16:15:57.869 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling - Modbus Error: [Input/Output] Error reading registers: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response

I'm not aware of getting any issues when using the HA built in modbus.

canton7, 2023-04-12:
Maybe it's not logging... Do you have a `delay` set in the built in modbus integration? Anything else which might be different?

iancg, 2023-04-12:
Yes - I'm getting some data in the UI - even for 11000

2023-04-12 16:19:30.221 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Refresh complete - notifying sensors: {11008, 11009, 11010, 11011, 11014, 11021, 11023, 11034, 11000, 11002, 11003, 11005}

iancg, 2023-04-12:
No delay:

    - name: FoxESSInverterModbus
      type: serial
      baudrate: 9600
      bytesize: 8
      method: rtu
      parity: N
      port: /dev/ttyUSB0
      stopbits: 1

I'll switch back and see if I can find the same logging for the built in one, unless there is anything else that I can try now?

canton7, 2023-04-12:
> Is the logging I'm copying only if all 3 retries fail? Or could I just be catching the failures that then are retried and succeed?

I'm not sure, to be honest.

canton7, 2023-04-12:
Are you in a position to be able to manually edit the files which make up this integration? They should be in `/config/custom_components/foxess`. If you edit `modbus_client.py` and change the final method to:

```py
    async def _async_pymodbus_call(self, call, *args):
        """Convert async to sync pymodbus call."""
        async with self._lock:
            result = await self._hass.async_add_executor_job(call, *args)
            await asyncio.sleep(30 / 1000)
            return result
```

... does that improve anything?

iancg, 2023-04-12:
If it becomes an issue I may just switch to the PoE version of the waveshare or direct lan.  The main reasons I went for the USB adaptor was that I didn't want another power supply and the firmware update had just blocked the lan route.

Yes - I can edit, will give that a go and report back.

canton7, 2023-04-12:
I mean, serial adapters should work (they work with the built-in modbus integration after all), so it would be good to fix them for everyone!

I suspect that delay might be relevant, since it always seems to be the not-first read which fails...

In theory you should be able to get some nice big reliable reads with a serial adapter as well...

canton7, 2023-04-12:
(Thanks a lot for your help here -- I don't have a serial adapter and my inverter's located such that it would be quite awkward to connect one up)

iancg, 2023-04-12:
The sleep appears to have fixed it:

    2023-04-12 16:51:43.081 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Refresh complete - notifying sensors: {11008, 11009, 11010, 11011, 11043, 11044, 11014, 11021, 11023, 11024, 11025, 11000, 11001, 11002, 11003, 11004, 11005}
    2023-04-12 16:51:43.200 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Refresh complete - notifying sensors: {11008, 11011, 11014, 11049, 11021, 11022, 11023, 11024, 11034, 11000, 11002, 11003, 11004, 11005, 11007}
    2023-04-12 16:51:52.095 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11000, 26)
    2023-04-12 16:51:52.096 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11000, 26, (247))
    2023-04-12 16:51:52.101 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11000, 26)
    2023-04-12 16:51:52.101 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11000, 26, (246))
    2023-04-12 16:51:52.247 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11034, 16)
    2023-04-12 16:51:52.247 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11034, 16, (247))
    2023-04-12 16:51:52.397 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11034, 16)
    2023-04-12 16:51:52.398 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11034, 16, (246))
    2023-04-12 16:51:52.528 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11069, 18)
    2023-04-12 16:51:52.529 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11069, 18, (247))
    2023-04-12 16:51:52.656 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11069, 18)
    2023-04-12 16:51:52.657 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11069, 18, (246))
    2023-04-12 16:51:52.795 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (41000, 12)
    2023-04-12 16:51:52.795 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (41000, 12, (247))
    2023-04-12 16:51:52.923 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (41000, 12)
    2023-04-12 16:51:52.924 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (41000, 12, (246))
    2023-04-12 16:51:53.044 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Refresh complete - notifying sensors: {11008, 11010, 11011, 11043, 11044, 11014, 11021, 11023, 11000, 11001, 11002, 11003, 11004, 11005}
    2023-04-12 16:51:53.170 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Refresh complete - notifying sensors: {11008, 11009, 11035, 11011, 11045, 11014, 11046, 11021, 11022, 11023, 11000, 11002, 11003, 11005, 11006}

canton7, 2023-04-12:
Amazing, thank you!

Can you test with 50-register reads?

canton7, 2023-04-12:
(Opened #143)

iancg, 2023-04-12:
128 at 5s

    2023-04-12 16:58:45.140 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11000, 87)
    2023-04-12 16:58:45.140 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11000, 87, (247))
    2023-04-12 16:58:45.140 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11000, 87)
    2023-04-12 16:58:45.141 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11000, 87, (246))
    2023-04-12 16:58:45.416 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (41000, 12)
    2023-04-12 16:58:45.417 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (41000, 12, (247))
    2023-04-12 16:58:45.696 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (41000, 12)
    2023-04-12 16:58:45.697 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (41000, 12, (246))
    2023-04-12 16:58:45.816 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Refresh complete - notifying sensors: {11008, 11009, 11010, 11011, 11014, 11015, 11021, 11022, 11023, 11024, 11025, 11034, 11035, 11036, 11037, 11038, 11041, 11042, 11043, 11044, 11045, 11046, 11048, 11049, 41000, 41001, 41002, 41003, 41004, 41005, 41006, 41009, 41010, 41011, 11069, 11070, 11071, 11072, 11073, 11074, 11075, 11076, 11077, 11078, 11079, 11080, 11081, 11082, 11083, 11084, 11085, 11086, 11000, 11001, 11002, 11003, 11004, 11005, 11006, 11007}
    2023-04-12 16:58:45.964 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Refresh complete - notifying sensors: {11008, 11009, 11010, 11011, 11014, 11015, 11021, 11022, 11023, 11024, 11025, 11034, 11035, 11036, 11037, 11038, 11041, 11042, 11043, 11044, 11045, 11046, 11048, 11049, 41000, 41001, 41002, 41003, 41004, 41005, 41006, 41009, 41010, 41011, 11069, 11070, 11071, 11072, 11073, 11074, 11075, 11076, 11077, 11078, 11079, 11080, 11081, 11082, 11083, 11084, 11085, 11086, 11000, 11001, 11002, 11003, 11004, 11005, 11006, 11007}
    2023-04-12 16:58:50.141 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11000, 87)
    2023-04-12 16:58:50.141 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11000, 87, (247))
    2023-04-12 16:58:50.142 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (11000, 87)
    2023-04-12 16:58:50.142 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (11000, 87, (246))
    2023-04-12 16:58:50.419 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (41000, 12)
    2023-04-12 16:58:50.419 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (41000, 12, (247))
    2023-04-12 16:58:50.696 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses for (41000, 12)
    2023-04-12 16:58:50.697 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Reading register: (41000, 12, (246))
    2023-04-12 16:58:50.829 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Refresh complete - notifying sensors: {11008, 11009, 11011, 11021, 11023, 11000, 11002, 11003, 11005, 11007}
    2023-04-12 16:58:50.953 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Refresh complete - notifying sensors: {11008, 11009, 11010, 11011, 11035, 11045, 11014, 11046, 11021, 11022, 11023, 11034, 11000, 11002, 11003, 11005, 11006, 11007}

canton7, 2023-04-12:
Nice! I'll use that size read then :)

Thanks again for your time!

closed 2023-04-22 by canton7