nathanmarlor/foxess_modbus #146: Period 2 - Start (Inner) etc doesn't update after Writing to registers

iancg, 2023-04-12:
## Version of the custom_component

Git head as of 5pm on 12th April

## Configuration

```yaml
Add your logs here.
```

## Describe the bug

After executing the following service call:

    service: foxess_modbus.write_registers
    data:
      start_address: 41001
      inverter: Inner
      values: 0,0,0,0,0,0

I see the entry in the log, and the inverter updates it's value, however the value of the sensor "Period 2 - Start (Inner)" etc doesn't reflect the modified value, but keeps the previous value.

[image]

Restarting HA doesn't cause the value to change, but using the FoxCloud app to set a new value does reflect.

## Debug log

```text

2023-04-12 19:32:32.308 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Writing register: (41001, ['0', '0', '0', '0', '0', '0'], 246)
2023-04-12 19:44:27.522 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Refresh complete - notifying sensors: {11008, 11009, 11010, 11011, 11014, 11021, 11022, 11023, 11024, 11034, 11035, 11037, 11045, 11046, 11049, 41005, 41006, 11000, 11002, 11003, 11006, 11007}
2023-04-12 20:36:54.137 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Writing register: (41001, ['0', '0', '0', '0', '0', '0'], 246)

```

canton7, 2023-04-12:
That's intentional. On modbus, you disable the charge window by setting the start and end times to midnight. Our integration adds a new sensor "Period X - Enable charge from grid" which says whether the charge window is enabled/disabled, and when disabled the start/end times remember the last time that was set.

This will be used to drive the lovelace card (when I get around to finishing it), so you can toggle the charge window off/on, and it doesn't forget the start/end times when you turn it off.

iancg, 2023-04-12:
Ah, OK, I was just trying this one:

    service: foxess_modbus.update_all_charge_periods
    data:
      inverter: "Inner"
      charge_periods:
        - enable_force_charge: false
          enable_charge_from_grid: false
        - enable_force_charge: false
          enable_charge_from_grid: false

Which gave a slightly different log:

2023-04-12 21:03:17.478 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_client] Writing register: (41001, [0, 0, 0, 0, 0, 0], 246)

But the result is the same - so I assume that now I can never see a time period of 00:00:00-00:00:00 again unless I set the enable_force_charge: true and set the times (to zero & zero)?

iancg, 2023-04-12:
> Note, there are 2 services to set charge windows, which do a bunch of validation to make sure you don't write anything invalid. It's not recommended to use the write_registers service to write these directly.

Yeah - I'd seen those, but was doing the "easy" conversion first from the old modbus automations.

iancg, 2023-04-12:
I'm still not convinced this is working.

If I go into FoxCloud and set a Enable force charge for period 2 of say 21:18 to 23:59 (used different times each time), but Enable charge from grid as off, then I see it update on the HA UI within a few seconds, but even with a service call of 

    service: foxess_modbus.update_all_charge_periods
    data:
      inverter: "Inner"
      charge_periods:
        - enable_force_charge: false
          enable_charge_from_grid: false
        - enable_force_charge: false
          enable_charge_from_grid: false

I then force close the FoxCloud app and re-open it, and the time period is still shown as 21:18-23:59.

iancg, 2023-04-12:
I wonder if I've managed to get the internal state of the component and the inverter out of whack.  Changing the time in FoxCloud seems to make the component come back into sync.  I'll change all my write_registers to update_all_charge_periods and see if that helps.

canton7, 2023-04-12:
Looks like they must have improved the app - it used to be that it would forget the times and revert to midnight if the charge period was disabled and you even backed out of the battery configuration page.

canton7, 2023-04-12:
Remember that the inverter itself loses the start and end times when you disable the charge period. So if something (app or HA) is showing you a start/end time when the charge period is disabled, its because its remembering the last start/end time it saw. The app only updates every 5 mins or so (writes happen straight away, but reads are slow)

canton7, 2023-04-12:
Using the service won't affect how the sensors in HA work here. It just does time to register conversion, validation, and let's you set of charge period at a time if you want.

iancg, 2023-04-12:
Ok - I think what might have been throwing me is that some of my automations used "start == 0" as a proxy for "enable_force_charge == false" and setting them to zero, didn't actually set them to zero.

The service is nicer.

BTW, is it intentional that they appear as "HH:MM:SS" in the sensor, but the service example just has "HH:MM" (no :SS)?

canton7, 2023-04-12:
We just return a python `time` from the sensor, and HA formats it with seconds. I'm not sure whether it's nicer to do that, or return a string.

The service lets you either include or omit seconds (but moans if they're non-zero)

canton7, 2023-04-12:
> Ok - I think what might have been throwing me is that some of my automations used "start == 0" as a proxy for "enable_force_charge == false" and setting them to zero, didn't actually set them to zero.

Yeah the intention is to hide the awkwardness of the inverter and provide the same interface as the app. So we've got sensors for enable force charge / enable charge from grid (corresponding to the two toggles in the app), then separate start and end. The services then provide the same interface

iancg, 2023-04-13:
I'm happy that it was just my expectations of previous behaviour + time lag in updating.  Hence closed issue.

Anyone migrating from the old modbus to this integration needs to change anywhere the have checked period start/end for 0 to instead check the boolean_sensor instead.

closed 2023-04-13 by iancg

canton7, 2023-04-13:
Great, thanks!