nathanmarlor/foxess_modbus #160: Polling settings reset after updating

screenagerbe, 2023-04-17:
## Version of the custom_component

1.4.0

## Configuration

n/a

## Describe the bug

When updating from 1.4.0b7 to 1.4.0 my polling settings were reset to the default values.
Might have been on purpose, but can't remember reading it in the release notes

## Debug log

n/a

canton7, 2023-04-17:
I don't think we've touched anything in this area between those versions...

screenagerbe, 2023-04-17:
I feel a bit awkward, but when updating my PROD container the settings were untouched.
So don't really know what was happening, might have been a glitch in TST related to me testing the different FoxESS integrations

canton7, 2023-04-18:
I tried to reproduce this and couldn't I'm afraid. Installed 1.4.0b7, set up a new inverter from scratch, customised the poll rate and max read, upgraded to 1.4.0, and my settings are still present (and are being used correctly).

screenagerbe, 2023-04-18:
I'll close this, I only encountered it in my "contaminated" TST container.
and it was not reproducible, so it's probably related to my environment.

closed 2023-04-18 by screenagerbe

canton7, 2023-04-18:
Cool, thanks! The config system in HA is a bit opaque, so I wouldn't be surprised if there was some hard-to-reproduce case which messed it up...