nathanmarlor/foxess_modbus #165: How do you set this up?

TehWardy, 2023-04-19:
This probably seems like a stupid question but look at the readme on the main page ...

[image]

Goto HACS, download using this github repo's main page as the link.
All good.

So you get it downloaded then then it goes "Configuration and Options", here's some UI pick your connection type.
Where is that?

Clicking on the downloaded integration in HACS just gives you the readme, am I missing something stupid here?

FozzieUK, 2023-04-19:
@TehWardy Once you’ve downloaded it through HACS you’ve added it to the list available, you then have to go to Settings, Devices & Services and click on the + Add Integration button. Search for Foxess modbus, click on it and it will take you to the setup page above.

TehWardy, 2023-04-19:
Legend ... 
I feel that would be useful in the main page guide if the devs don't mind adding this. 
Thanks @FozzieUK  :)

closed 2023-04-19 by TehWardy

canton7, 2023-04-19:
Yeah, I want to turn the wiki into a proper step-by-step guide at some point, but it becomes a matter of free timee

screenagerbe, 2023-04-20:
What if you simply add

```
5. Restart Home Assistant
6. Go to Settings > Devices and Services > Add Integration 
7. Search for and select 'FoxESS - Modbus' (If the integration is  not found, empty your browser cache and reload the page).
8. Proceed with the configuration
```
below this:
[image]

I'm away this weekend, but I could create a PR for such changes (got no python knowledge, so I might be able to make myself useful in another area) 🙂

FozzieUK, 2023-04-23:
Thanks, PR added #178