nathanmarlor/foxess_modbus #172: Energy Dashboard - unable to edit after automatic setup

ryaaanmorgan, 2023-04-22:
[image]

After setting up the energy dashboard automatically through the integration, it appears you can no longer make any changes to it. 

In my case, I tried to add the solcast forecast into the PV production bit and it wouldn't allow it. 

Nath has pointed me in the direction of being able to reset it manually but might be worth considering how to do this automatically?

Thanks for all the hard work on this integration.

canton7, 2023-04-22:
The image is a bit too compressed to see -- does that say "Unknown error" at the top?

canton7, 2023-04-22:
So, I think the workaround is to click the "Edit" button on "Grid Consumption Total", then click "Save". 

It looks like the energy dashboard stuff is expecting us to have set `entity_energy_price` (which we never did), and it's falling over because of it. Editing/saving "Grid Consumption Total" fixes this.

I'll look at getting this fix in automatically.

canton7, 2023-04-22:
That said, the docs for `entity_energy_price` imply it doesn't have to be set...

canton7, 2023-04-22:
Ah right, but the energy prefs stuff uses TypedDict, and there's a difference between leaving a field unset and explicitly setting it to None, and HA is assuming we've done the latter. I think this is fixable in a migration though. I'm going to get #115 in quickly first, then add this on top.

closed 2023-04-22 by canton7

canton7, 2023-04-22:
That should be fixed in the latest beta, [v1.4.1b1](https://github.com/nathanmarlor/foxess_modbus/releases/tag/v1.4.1b1). Just upgrading should be enough to fix it -- you won't need to remove/re-add.

Please let me know if it doesn't work for you.

ryaaanmorgan, 2023-04-22:
The fix worked perfectly. Thank you!!!!

PR #174 (2023-04-22): Fix energy dashboard
It turns out that the energy dashboard wants us to specify all keys, even if their value is None. However, that's not what TypedDict does by default.

Fix the setup in the config flow, and add a migration to fix existing breakages.

Fixes: #172
commit f704970d (2023-04-22): Fix energy dashboard

It turns out that the energy dashboard wants us to specify all keys,
even if their value is None. However, that's not what TypedDict does by
default.

Fix the setup in the config flow, and add a migration to fix existing
breakages.

Fixes: #172