nathanmarlor/foxess_modbus #176: Read software versions / serial number on startup?

canton7, 2023-04-22:
Not sure whether we need to / where we'd display them.... I don't really want to poll them all the time, as they're unlikely to change.

canton7, 2023-04-22:
See https://github.com/TonyM1958/HA-FoxESS-Modbus/issues/5

TonyM1958, 2023-04-23:
for what it's worth, I added scan_interval_static for loading model number, serial number, inverter firmware versions and battery firmware version info to my secrets and set this to 3600 so they just get scanned once an hour. Thought this was a compromise that would load the values at startup and not put a significant on-going load on the system but reflect changes in versions.

canton7, 2024-04-28:
We now read software versions on start-up.

closed 2024-04-28 by canton7