nathanmarlor/foxess_modbus #192: Lost connection to inverter after changing settings multiple times in quick succession, needs HA restart

canton7, 2023-04-25:
From https://github.com/nathanmarlor/foxess_modbus/discussions/40#discussioncomment-5720147:

> The only issue was the one I mentioned with going in/out of the config repeatedly to change poll settings - after 3 or 4 times in short succession it goes into error (see below), I tried deleting the integration and adding again but it failed as it couldn't access the IP address - it needed an HA restart to resolve.

> These were the logs when it lost it:-

```
2023-04-25 12:04:08.819 DEBUG (SyncWorker_6) [custom_components.foxess_modbus.modbus_client] Connecting to ModbusBaseClient._params(host='192.168.0.238', port=502, framer=<class 'pymodbus.framer.socket_framer.ModbusSocketFramer'>, timeout=3.0, retries=3, retry_on_empty=False, close_comm_on_error=False, strict=True, broadcast_enable=False, kwargs={'modbus_type': 'TCP'}, reconnect_delay=100, reconnect_delay_max=300000, baudrate=None, bytesize=None, parity=None, stopbits=None, handle_local_echo=None, source_address=None, sslctx=None, certfile=None, keyfile=None, password=None, server_hostname=None)
2023-04-25 12:04:08.824 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling tcp://192.168.0.238:502 247 - Modbus Error: [Input/Output] Error reading registers from tcp://192.168.0.238:502: Modbus Error: [Input/Output] [Errno 104] Connection reset by peer
2023-04-25 12:04:14.817 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.0.238:502 247: (31000, 25)
2023-04-25 12:04:14.820 DEBUG (SyncWorker_2) [custom_components.foxess_modbus.modbus_client] Connecting to ModbusBaseClient._params(host='192.168.0.238', port=502, framer=<class 'pymodbus.framer.socket_framer.ModbusSocketFramer'>, timeout=3.0, retries=3, retry_on_empty=False, close_comm_on_error=False, strict=True, broadcast_enable=False, kwargs={'modbus_type': 'TCP'}, reconnect_delay=100, reconnect_delay_max=300000, baudrate=None, bytesize=None, parity=None, stopbits=None, handle_local_echo=None, source_address=None, sslctx=None, certfile=None, keyfile=None, password=None, server_hostname=None)
2023-04-25 12:04:14.823 ERROR (SyncWorker_2) [pymodbus.logging] Connection to (192.168.0.238, 502) failed: [Errno 104] Connection reset by peer
2023-04-25 12:04:14.831 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling tcp://192.168.0.238:502 247 - Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.0.238:502)]
2023-04-25 12:04:20.818 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Reading addresses on tcp://192.168.0.238:502 247: (31000, 25)
2023-04-25 12:04:20.820 DEBUG (SyncWorker_0) [custom_components.foxess_modbus.modbus_client] Connecting to ModbusBaseClient._params(host='192.168.0.238', port=502, framer=<class 'pymodbus.framer.socket_framer.ModbusSocketFramer'>, timeout=3.0, retries=3, retry_on_empty=False, close_comm_on_error=False, strict=True, broadcast_enable=False, kwargs={'modbus_type': 'TCP'}, reconnect_delay=100, reconnect_delay_max=300000, baudrate=None, bytesize=None, parity=None, stopbits=None, handle_local_echo=None, source_address=None, sslctx=None, certfile=None, keyfile=None, password=None, server_hostname=None)
2023-04-25 12:04:20.823 ERROR (SyncWorker_0) [pymodbus.logging] Connection to (192.168.0.238, 502) failed: [Errno 104] Connection reset by peer
2023-04-25 12:04:20.827 DEBUG (MainThread) [custom_components.foxess_modbus.modbus_controller] Modbus exception when polling tcp://192.168.0.238:502 247 - Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.0.238:502)]
```

canton7, 2023-04-25:
I think I might have seen this as well using UDP... I put it down to the W610 not liking connection from multiple sources at the same time, but it might actually be this (with a slightly different message of course, because UDP).

FozzieUK, 2023-05-17:
@canton7 yeah happened to me again last night, I only made a single change this time and it lost it, sounds like something in the underlying python libraries is getting messed up. For the average user it won’t happen very often and a restart fixes it so perhaps keep this on the back burner.

canton7, 2023-05-17:
I'm not sure if it can be pymodbus -- the packets it's sending out look fine. It's just that (in my case) the adapter wasn't responding

canton7, 2023-05-17:
Well, I'm looking at UDP and the W610 -- there's no socket state there.

canton7, 2023-05-28:
I noticed that the TCP connection to the inverter/adapter wasn't being properly closed when the integration was reloaded, which might be related: I can imagine lots of not-yet-timed-out TCP sockets causing the adapter/inverter some issues

FozzieUK, 2023-05-28:
Ok, fingers crossed that's what it is - ironically last week I spent over 10 minutes going in / out of the config changing the number of reads, poll time etc.. and it didn't break once :persevere: :smirk: