nathanmarlor/foxess_modbus #193: Battery charging to full every night after switching to modbus

andykorol1, 2023-04-26:
This error originated from a custom integration.

Logger: custom_components.foxess_em.fox.fox_modbus
Source: custom_components/foxess_em/fox/fox_modbus.py:94
Integration: FoxESS - Energy Management (documentation, issues)
First occurred: 6:30:12 AM (1 occurrences)
Last logged: 6:30:12 AM

Error writing holding register - retry (1/5): Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response

This error originated from a custom integration.

Logger: custom_components.foxess_modbus.modbus_controller
Source: custom_components/foxess_modbus/modbus_client.py:77
Integration: FoxESS - Modbus (documentation, issues)
First occurred: 2:25:23 AM (5 occurrences)
Last logged: 6:30:22 AM

General exception when polling tcp://192.168.1.234:502 247 - AttributeError("'WriteMultipleRegistersResponse' object has no attribute 'registers'")
General exception when polling tcp://192.168.1.234:502 247 - AttributeError("'WriteSingleRegisterResponse' object has no attribute 'registers'")
Traceback (most recent call last):
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 136, in _refresh
    reads = await self._client.read_registers(
  File "/config/custom_components/foxess_modbus/modbus_client.py", line 77, in read_registers
    return response.registers
AttributeError: 'WriteMultipleRegistersResponse' object has no attribute 'registers'

I'm getting a few errors in the log as above and my battery seems to be charging to full each night even though the foxess em says 0 charge required. 

## Version of the custom_component

## Configuration

```yaml
Add your logs here.
```

## Describe the bug

A clear and concise description of what the bug is.

## Debug log

```text

Add your logs here.

```

andrewmooreio, 2023-04-26:
Are you connecting directly to the ethernet port on the inverter, or are you using an RS485 to ethernet adapter?

andykorol1, 2023-04-26:
RS485 adapter

canton7, 2023-04-26:
Which adapter?

canton7, 2023-04-26:
Also, the integration won't affect battery charging - this isn't foxess_em

andykorol1, 2023-04-26:
Waveshare POE

canton7, 2023-04-26:
Interesting.

It looks like foxess_em sent a write request (presumably to change the charge period) and foxess_modbus send a read request. However, foxess_em didn't get a response, and foxess_modbus got the response which was intended for foxess_em.

So foxess_modbus fell over, because it issued a read request but the response it got back was for a write request (I'm surprised that pymodbus didn't pick this up), and presumably it's also related to why foxess_em wasn't able to set the charge period properly. This is probably a problem with the Waveshare itself -- it's returning the wrong response to the wrong client.

Did you enable "Enable RS485 Multi-Host" in the settings, and detailed [in our instructions](https://github.com/nathanmarlor/foxess_modbus/wiki/Waveshare-RS485-to-ETH-%28B%29)?

andykorol1, 2023-04-26:
So I have enabled RS485 multi host and the errors seem to have gone quiet now. Will see what happens over night. Strangely I never found the instructions that you linked to before now, all the same except the multi host so hopefully they won't clash now.

canton7, 2023-04-26:
Good to hear! The instructions will be linked from the setup wizard in the next release of this integration, but it's all still a work in progress.

It's weird that the waveshare doesn't enable this by default, as I can't imagine you'd ever want the alternative, but oh well.

canton7, 2023-05-04:
As an action, we should check that pymodbus is giving us back the message type that we expect, and throw a more useful error if it doesn't

PR #241 (2023-05-17): Log a more useful warning if we get the wrong response type back from pymodbus
Fixes: #193
commit 6754a3a7 (2023-05-17): Log a more useful warning if we get the wrong response type back from pymodbus

Fixes: #193
closed 2023-05-17 by canton7