nathanmarlor/foxess_modbus #197: Unknown model connecting to H3

canton7, 2023-04-26:
Thank you for the first beta integration, when adding it to HA i'm receiving the following Info: 
[image]

H3 Inverter - USR-610

_Originally posted by @benbeton in https://github.com/nathanmarlor/foxess_modbus/issues/167#issuecomment-1523852010_

canton7, 2023-04-26:
@benbeton can you look in your log (Settings - System - Logs) for warning messages from foxess_modbus? There should be one saying what model the inverter reported itself as.

benbeton, 2023-04-26:
Log: 
[image]

canton7, 2023-04-26:
Oh the bugger, it's started it with a space. Right, I'll try and get a fix in later tonight

canton7, 2023-04-26:
Actually, I assume that's a space. Can you just use modpoll to read holding register 30000 for me? (with `-t 4`)

benbeton, 2023-04-26:
is the something wrong with my cmd? 
[image]

canton7, 2023-04-26:
I think you're reading too many registers (and some of the later ones are invalid) -- just `-c 5` or so.

canton7, 2023-04-26:
Try v1.4.1b4

closed 2023-04-26 by canton7

benbeton, 2023-04-26:
that did the trick, works fine:  I can see all the values incoming
[image]

thank you, that's great!

canton7, 2023-04-26:
Awesome, thanks for confirming! Let me know if any of the values don't look quite right

PR #198 (2023-04-26): Fix auto-detection of H3 inverters
Their model seems to start with a space.

Fixes: #197
commit 3830c132 (2023-04-26): Fix auto-detection of H3 inverters

Their model seems to start with a space.

Fixes: #197