nathanmarlor/foxess_modbus #199: Simple application that returns json

kalwinskidawid, 2023-04-27:
Hi, are you planning or do you have somewhere a working version that would allow you to read data from Foxess, but bypassing HA. 
So that it could be fired straight from the python interpreter?

canton7, 2023-04-27:
No, no plans and there likely won't be any.

We're very tightly integrated with HA: we read register values over modbus and turn them into human units, then send that to HA which handles all of the formatting and display; all of the setup and configuration is handled by HA; the integration sensors (total PV1, etc) are implemented in HA; the storage we use for the charge period sensors is handled by HA; etc etc. See [e.g. here](https://github.com/nathanmarlor/foxess_modbus/blob/55dc742da2be8cebf19a34a1cb398cbd5c84923e/custom_components/foxess_modbus/entities/entity_descriptions.py).

You might be able to create a tool which uses the entity definitions to parse the registers it reads, but that will depend on HA.

You can read registers directly from the command-line using e.g. modpoll or pymodbus, but then you'll need to do your own parsing.

kalwinskidawid, 2023-04-27:
It's a pity , I'm just trying to decode it to a simpler form. Unfortunately, not knowing the HA operating scheme is quite problematic at this point. Anyway, I'm mega grateful for running this project, at least I know it's possible.

closed 2023-05-03 by nathanmarlor