nathanmarlor/foxess_modbus #213: Battery Charge Status missing after switching from Modbus

tekhiegit, 2023-05-03:
Hi, I've been using the old method of hard writing into the config.yaml and modbus.yaml to read from my H1.5E inverter for a while and have recently switched over to this much cleaner implementation, but it's missing one of the g features that I used the most previously and I can't figure out how to get it back (or if I'm just missing it)

There used to be a sensor called battery-charge-status which showed as a positive number when charging the batteries or a negative number when discharging. I used this with gauges to instantly see what was going on with energy use in a live way. It easily let me see if my house load was greater than my solar generation etc, and if the battery was depleted it would just show as zero. 

It's it possible to bring back this sensor, or is there some way I can add it in myself?

Cheers.

canton7, 2023-05-03:
Maybe it was Battery Discharge Power? In which case that's now "Inverter Battery Power" (invbatpower)

tekhiegit, 2023-05-03:
Think I was using version 0.9 not sure which implementation, I hacked it in when I first starting using HA, I didn't even know about using HACS 🙄. 

The setting above still give a positive number with battery drainage.

canton7, 2023-05-03:
I don't think there's ever been a H3 sensor with "status" in the name, looking through the history.

"Inverter Battery Power" should be signed. Battery Charge will be non-zero when Inverter Battery Power is negative, and Battery Discharge will be non-zero when Inverter Battery Power is positive.

Both the old Battery Discharge Power and the new Inverter Battery Power are positive for discharging, negative for charging. That matches what you're seeing, no?

What values do you get when you're charging?

Maybe you created a template sensor in your configuration.yaml which flips the sign?

FozzieUK, 2023-05-04:
@tekhiegit I've never known there to be a charge status sensor included as part of the old modbus, 

I did write a template sensor some time ago to do similar (and probably shared it on the facebook group)

First check you configuration.yaml to make sure there isn't a template sensor already defined, if there is to get it working just change the source sensor names as follows - sensor.discharge_power to sensor.discharge and sensor.charge_power to sensor.charge

If it isn't there add this to your configuration.yaml templates it will give you a sensor in watts that is positive for charge, negative for discharge. 

```
      battery_charge_status:
        friendly_name: "Battery Charge or Discharge"
        unit_of_measurement: "W"
        value_template: "{{ ( (states('sensor.battery_charge') | float(default=0) - states('sensor.battery_discharge') | float(default=0)) * 1000) | round(0) }}"
```

If you want it in kW change the * 1000 to * 1, the unit back to "kW" and the round(0) to round(2)

canton7, 2023-05-04:
Aha, maybe that explains it.

Note that the old Battery Discharge Power (and the new Inverter Battery Power) are the same as battery_charge_status, but with the sign flipped (battery_charge and battery_discharge are derived from Battery Discharge Power, after all).

tekhiegit, 2023-05-05:
> @tekhiegit I've never known there to be a charge status sensor included as part of the old modbus,
>
> I did write a template sensor some time ago to do similar (and probably shared it on the facebook [...]

Ah, Amazing! Yes thank you, it was all done so long ago and when I had basically zero yaml or HA knowlage, so when I added in this modbus I just stripped everything out from config, I was clearly too brutal with the delete key. Thank you both for you help.

closed 2023-05-05 by tekhiegit