nathanmarlor/foxess_modbus #223: Individual Pollrates per address-block

rsaemann, 2023-05-07:
Problem: I need some of the sensors in very fast update intervall for near-real-time automations, some in a medium-update intervall for information and some only very rarely for error-tracking and statistics.
Polling all of them with a very high frequency blocks the polling and so many write calls to the database may damage the SD/SSD.

Solution: **Different poll-intervall for address blocks delivering the same type of content.** 
For example:
* 10 seconds for PV power + house load power + battery power
* 60 seconds for Battery temperature, SOC, current, voltage...
* 300 seconds for daily/total energy 

Alternatives: Disabling the rarely read sensors. And install another instance on the same address with another poll rate. (Not tried)

nathanmarlor, 2023-05-07:
Duplicate of https://github.com/nathanmarlor/foxess_modbus/discussions/71

canton7, 2023-05-07:
The thing is, with any decent adapter (ie not the W610), we can read everything in two reads (one for the 10000 range, one for the 4000p range).

The coat of sending a request (of any size) is much much higher than the cost of making the request larger. 

So it's much more efficient to do 2 large requests and read everything, than start splitting things up, and paying the overheads of loads of separate requests.

canton7, 2023-05-07:
Also, we only tell HA that an entity has changed if its value has actually changed. So if you're polling things which aren't changing, that's literally free.

canton7, 2023-05-07:
> Polling all of them with a very high frequency blocks the polling 

I don't think this is true (at least if you have a half decent adapter). Do you have any evidence for this?

nathanmarlor, 2023-05-07:
@canton7 this is true for the native integration, but not this one (as we've both worked to ensure 😀). Perhaps we need to add to the readme explaining how this integration operated and why it's different/better suited?

canton7, 2023-05-07:
Yeah, the readme mentions it tangentially but we could make it clearer for sure!

canton7, 2023-05-07:
I created an FAQs wiki page, so that might be a good spot too

canton7, 2023-05-12:
Closing as no response. Please re-open as a discussion if you have any more questions!

closed 2023-05-12 by canton7