nathanmarlor/foxess_modbus #233: Investigate/add new sensors

canton7, 2023-05-13:
Now that we have a proper modbus reference document, there are some new registers we might consider:

 - SubsystemNumber, 11047
 - BatMinSoc, 11050
 - BatChargeRequest, 11051
 - BatChargeLimit, 11052
 - BatDischargeLImit, 11053
 - BatModuleType, 11054
 - Inverter state, 11056
 - BatStatus, 11057
 - BMS connect state, 11058
 - Meter1 connect state, 11059
 - Meter2 connect state, 11069
 - Fault codes

And then there's max/min charge current

TonyM1958, 2023-05-14:
Worth noting:

+ Subsystem number is the BMS Battery Count
+ BatMinSoc is just the BMS version of Battery SoC, it's not min at all. It may just be the value passed from the BMS to the inverter to display on the front panel and then replayed on the inverter modbus.
+ BatStatus is really BMS Status (Normal, Offline etc)
+ Bat Module Type - not clear what this represents. I get the value 84. Could relate to the BMS or the battery? Really need people to contribute different values to see what range we get with different equipment.

Codes / decodes are now in the H1 H3 Modbus map spread sheet on the Codes tab. For info - in general, where things use a code, I have ended the modbus sensor with _code and created a template that decodes the value so it is readable and can be displayed. The decoded entity names are now included in the spread sheet.

Also look at Meter 1 Code (40010) and Meter 2 Code (40011) - these are the Meter / CT / Disabled setting is.
Grid Code (40006) is the G98 / G99 setting on the inverter, probably worth picking up as well but you will need to work out more of the codes for different regions / regulations.

Fault codes - word / bit map also added to the spread sheet. All of the fault codes seem to map to bits, but there are not many Fault Codes available, as recorded on the Fox ESS cloud.  The PDF I found on AWS relates to H3 fault codes, not H1. Also note that BMS Lost and BMS External Fault seem to be different messages for the same fault.

Max / min charge and discharge currents are worth a look at. The front panel, settings are 41007/41008 but these are over-ridden by the BMS values at 11041/11042. For example, if Max Charge Current is 30A at 41007, but the BMS wants to restrict this to 5A due to the charge state of the battery, it sets the value at 11041 to 5A to reduce the current sent from the inverter. So, you might have a high max current setting, but the BMS can also refuse to accept this.

TonyM1958, 2023-05-14:
Here's some examples where I've thrown new sensors onto cards to see what they do. All the input register addresses I've tried so far from the PDF seem to work ok.

[image]

Battery limit flag seems strange. This value goes from 1281 to 1282 when the battery is full so it looks like only the bottom 2 bits are changing. The other bits might mean something...?

Charge and Discharge Cut Voltages are the BMS limits for where it will cut off the battery when charging or discharging to stop over charging or depleting the batteries too much. These values are calculated based on the number and type of battery, it appears (I have 6 x HV2600).

I've not seen any changes in Power Limit Flag, BMS Charge Request, BMS Charge Limit or Connect Time Count (s) so far. Possibly, the last one is a time-out that only counts if a connection fails. I've not max'd out my inverter, so don't expect to hit a power limit.

Ndrinta, 2023-05-30:
Hey @canton7,
may be worthwhile having a look at the scaling for `BMS Energy Throughput` while you're at it.
It seems little off to me that ~700kWh have gone through my battery today!

[image]

canton7, 2023-05-30:
@Ndrinta Fixed in 1.5.0, are you sure you're up to date? If that's not it, please open a new issue, as this issue is not related.

Ndrinta, 2023-05-30:
Ok, I'll delete this from here (BTW I'm on 1.5.2).

canton7, 2023-05-30:
Cheers! It does seem a bit high on my system too, but we can investigate im the new issue

Ndrinta, 2023-06-02:
Hey, what's BatChargeRequest?

Ndrinta, 2023-06-02:
I spent some time investigating faults. Address `11061` returns a U16 where:

Bit0 - GridLostFault
Bit1 - GridVoltFault
Bit2 - GridFreqFault (?)
Bit3 - Grid10minVoltFault
Bit4 - EpsVoltFault
Bit5-15 - ???

PS: Where is the list of all addresses we were talking about?

canton7, 2023-06-02:
The faults were implemented in v1.5.3b3.

See https://github.com/nathanmarlor/foxess_modbus/wiki/Protocol-Documents

FozzieUK, 2023-06-02:
@Ndrinta 
> Hey, what's BatChargeRequest?

it’s a signal from the BMS to the inverter to turn the charger on.  It’s a very rare occurrence to see it as the inverter usually drives all the charging events, but if the BMS has a low temp & low soc or needs balancing charge it will request that the inverter turns it’s charger on (usually with a low charge rate set)

closed 2024-04-28 by canton7