nathanmarlor/foxess_modbus #245: Update Charge Period 2

FozzieUK, 2023-05-16:
I've been using my own charge period setting, but just tried the integration built in service call - when I set Charge Period 2, it always sets Charge Period 1 instead, and if I set Charge Period 1 it does indeed set Charge Period 1.

The service set as this -

[image]

Results in Charge period 1 being set, and period being left as it was

[image]

There are no error logs with default 'warn' logging set

Be grateful if you could take a look - I am on the latest beta (despite your warning not to use 🙄 ), I can fall back to 1.5 if you prefer me to test with that.

canton7, 2023-05-16:
Reproduced. Bugger, I must have broken that when refactoring how the sensors are represented. Thanks!

canton7, 2023-05-16:
Not only does it program 1 when it should program 2, it also copies 2 to 1. So you do get both periods set, but the opposite way around from what you were expecting! That leads to some confusion when you're trying to disable both charge periods...

FozzieUK, 2023-05-16:
I couldn’t quite work out what was going on, I had to keep clearing them before I could try again as it complained about overlaps - Thanks, let me know if you want me to test it 👍

canton7, 2023-05-16:
Thanks! I've got a day free tomorrow, so I'll get it tested, merged and released then

PR #246 (2023-05-17): Properly set the correct charge periods from the service
Currently if the user sets period 2, we'll program 2 to 1 and copy 1 to 2

Fixes: #245
commit bbfe957e (2023-05-17): Properly set the correct charge periods from the service

Currently if the user sets period 2, we'll program 2 to 1 and copy 1 to 2

Fixes: #245
closed 2023-05-17 by canton7