nathanmarlor/foxess_modbus #264: H1 State Sensor

gjhill6100, 2023-05-19:
Would it be possible to include the inverter State sensor ?

It would then be good to be able to use email notifications if the Inverter goes into particular states that need attention.

The other week, the main MCB for the Inverter tripped but this was not spotted for quite a while as the system does not automatically send any notifications of any system faults.

We could then maybe use the State sensor to generate some HA notifications.

canton7, 2023-05-20:
Yep, this is next on my list!

The only thing I'm a bit nervous about is that actually reading the fault code requires an 8-register read, which is going to add quite a bit of time to anyone using a W610 over TCP

calum-mcfarlane, 2023-05-22:
Can't help feeling that you're eventually going to have to add a disclaimer to the integration saying "The W610 isn't great, honestly" and setting the read frequency to 15 seconds or whatever.

closed 2023-05-27 by canton7

gjhill6100, 2023-05-28:
> completed via #280

Hi,

Is this all implemented in a new HACS version ?

Kind Regards

Graham

canton7, 2023-05-28:
It's out in the pre-release version v1.5.3b3

PR #280 (2023-05-27): Add fault codes and inverter states
Fixes: #264