nathanmarlor/foxess_modbus #299: "BMS Energy Throughput" seems off.

Ndrinta, 2023-05-30:


canton7, 2023-05-30:
Thanks! We did originally fix the scale on this, then I seem to have reverted it https://github.com/nathanmarlor/foxess_modbus/commit/482539436f493007c704a45822733af607e1dca2.

closed 2023-05-30 by canton7

TonyM1958, 2023-05-30:
FWIW, I tracked this for a while and its in watts, scale : 0.001. Its supposed to be the energy counter that drives BMS Cycle Count, but it is uint16 and wraps every 65kWh, making it hard to use for anything useful.

The metric that I use that makes more sense is a template Battery Energy Total. This is the sum of Charge Energy Total and Discharge Energy Total, similar to the above, but uint32 so it does wrap,and tracks battery use since install. As BMS Cycle Count also counts from installation, it then makes sense to track Battery Energy per Cycle  by dividing Battery Energy Total by BMS Cycle Count.

Ndrinta, 2023-05-30:
Great that will be fixed, thanks, but after seeing the discussion I believe I'll disable it at once.
I think we should close this, I'll let you the onor.

PR #300 (2023-05-30): Fix scale in bms_watthours_total
This was originally fixed in #168, but #167 seems to have accidentally reverted it.

Fixes: #299
commit c73901e3 (2023-05-30): Fix scale in bms_watthours_total

This was originally fixed in #168, but #167 seems to have accidentally reverted it.

Fixes: #299