nathanmarlor/foxess_modbus #309: Get us included in the hacs registry

canton7, 2023-06-05:
This should make life a bit easier for people

nathanmarlor, 2023-06-06:
Done! 🙂

canton7, 2023-06-06:
Just making sure you don't have any objection to getting us in the default HACS registry?

canton7, 2023-06-06:
Ah, looks lik you have to be the one to submit: https://hacs.xyz/docs/publish/include#check-owner

nathanmarlor, 2023-06-06:
No problem, I'll do it first thing tomorrow 👍

nathanmarlor, 2023-06-09:
@canton7 https://github.com/hacs/default/pull/1934

:)

canton7, 2023-06-09:
Awesome thanks! Looks like they're running at least a month behind, so now we wait

canton7, 2023-07-07:
@nathanmarlor Did you intend to add this commit? https://github.com/hacs/default/pull/1934/commits

nathanmarlor, 2023-07-07:
@canton7 it was an auto GitHub update against main thing - I'm only on my phone though, has it done something weird? Just checking in to see how if they'd merged us into HACS yet.

canton7, 2023-07-07:
It just wasn't clear what it was or why it was there. We seem to get one shot at being merged before they go AWOL for another few months, so I'm just a little concerned about anything wjodh might raise a question/comment, rather than an insta-merge

nathanmarlor, 2023-09-21:
@canton7 

[image]

😁😁😁😁😁

canton7, 2023-09-21:
Yaaaaaay! I've got some time tomorrow, will try and fix the two latest issues and update the readme.

commit 3abcd3ec (2023-09-30): Update instructions as we're now in hacs

Fixes: #309
PR #426 (2023-09-30): Update instructions as we're now in hacs
Fixes: #309
closed 2023-09-30 by canton7