nathanmarlor/foxess_modbus #334: Failed value validation against a rule

kozobera, 2023-06-15:
Hi Nathan, all,

I have AIO-H3-08, foxes Modbus v1.6.1, RUNCCI-YUN USB and I am receiving this error again and again (~100 per hour). 

Logger: custom_components.foxess_modbus.entities.modbus_entity_mixin
Source: custom_components/foxess_modbus/entities/modbus_entity_mixin.py:92
Integration: FoxESS - Modbus (documentation, issues)
First occurred: 06:52:58 (108 occurrences)
Last logged: 08:41:19

Value (-125: -0.125) for entity 'sensor.foxess_inv_power_s' address(es) '[31013]' failed validation against rule (Range : {'_min': 0, '_max': 100})
Value (-117: -0.117) for entity 'sensor.foxess_inv_power_s' address(es) '[31013]' failed validation against rule (Range : {'_min': 0, '_max': 100})
Value (-1: -0.001) for entity 'sensor.foxess_load_power_s' address(es) '[31030]' failed validation against rule (Range : {'_min': 0, '_max': 100})
Value (-2: -0.002) for entity 'sensor.foxess_load_power_s' address(es) '[31030]' failed validation against rule (Range : {'_min': 0, '_max': 100})
Value (-3: -0.003) for entity 'sensor.foxess_load_power_s' address(es) '[31030]' failed validation against rule (Range : {'_min': 0, '_max': 100})

In my case, an installation was done via HACS, a standard one, with no tweaking. In fact, all three phases are affected by this error (in this log, only phase S is shown).
[image]

All this results in missing data. See picture.

David

canton7, 2023-06-15:
In fairness, load can legitimately be negative (e.g. if there's an extra inverter on the house side of the CT clamp which the inverter doesn't know about), so I think we should lift that restriction across the board.

I'm not sure why inverter power is negative though. We do see small negative values on things like PV1/2 as well. Maybe we should just be clamping these to zero, rather than marking them invalid?

FozzieUK, 2023-06-15:
It's more of a problem with 3 phase as you get imbalances across the phases, that coupled with changing power factors can upset real time power readings, hence occasional negative values.

I would agree lift the restrictions, and clamp the inverter power to zero when negative. 👍

kozobera, 2023-06-16:
Why do you want to clamp the value to zero when negative? I believe, in real life, when the load is imbalanced, we can see small negative power in one phase. Nevertheless, the sum of all phases is always positive. Does a negative power somehow ruin subsequent calculations in the HA?

I would just ignore (meaning no range check) negative values ​​and use them normally.

David

canton7, 2023-06-16:
That's what we just agreed above.

The problemtthat's trying to catch is that the inverter can sometimes return nonsense values (particularly if it's booting, I think). So there's validation that values at least look vaguely as expected, which stops us from recording a lot of the rubbish (which throws things like stats way out)

kozobera, 2023-06-16:
> That's what we just agreed above (inverter output power clamped, load power can be negative)?

I see. It is clearer now. My misunderstanding. English is not my primary language, and I just learned another meaning of "clamped" :-)

closed 2023-06-28 by canton7

canton7, 2023-10-15:
Ack, inverter output power *can* be negative, if it's charging the battery.

PR #350 (2023-06-28): Fix H3 validation
Fixes: #334
commit 0c37926b (2023-06-28): Load powers can be negative for the H3 as well

Fixes: #334