nathanmarlor/foxess_modbus #357: Unable to change adapter model

allseven, 2023-07-02:
I've added a fourth EC4100 Cube to my stack and am checking the configuration of the integrations I'm using.

I've gone into FoxESS-Modbus and in 'Network' I have found the adapter model is set to 'Other' rather than 'USR-W610'. If I click on the correct 'USR-W610' and then 'Submit' twice I get an error 'You have already set up an inverter with this address' and I have to close the panel without making the change stick. Is there a way round this?

canton7, 2023-07-02:
Weird. Upgrade to the latest beta - I've reworked that bit of the config, and the adapter selection is now in configure - network settings.

allseven, 2023-07-02:
Thanks. I've upgraded to v1.7.0b1 but the problem persists.

canton7, 2023-07-03:
Under Devices & Services, does it say "1 Device" or "2 Devices" on the card?

allseven, 2023-07-03:
There's two devices. I have an SMA Sunny Boy 4000TL inverter from a 2011 installation and a Fox H1-5.0 hybrid from a parallel (to retain my FiT) 2022 installation. When I set Modbus up I configured it with both but realised later that the SMA side would probably be redundant as it could not be read directly so I disabled it. Hope that, and the image, helps.
[image]

canton7, 2023-07-03:
Interesting, I wonder whether that's messed up the duplicate detection somehow.

I'm away from a computer for the next week or two. The Other adapter is currently identical to the W610 so there's no harm in leaving things as they are for a bit. If you want to get it resolved, you can remove/readd the integration (which is the supported way to add / remove inverters), and your entitt history won't be lost provided you use the same entity id prefix.

allseven, 2023-07-03:
Thanks. I've deleted it and reinstalled it. All seems well.

canton7, 2023-08-14:
I wasn't able to replicate this I'm afraid, so I'm not sure what was going on. Closing, as there's nothing I can do. If you hit the same issue again, or if anyone else hits this, please say something and we can re-open.