nathanmarlor/foxess_modbus #371: Load power or math issue

zaapp, 2023-07-20:
I wonder why the Load Power R/S/T entities have stopped working. They worked for the first two days of using the integration.
To illustrate the situation:
Row one -> Grid consumption
Row two -> Feed-in
Row three -> Inverter power
Row four -> Load (doesn't work, doesn't agree with what I feed into the grid and the inverter production.
Fifth row -> From battery
[image]

canton7, 2023-07-20:
Looks like it might be the same as #362. 

We haven't changed anything on our side, and indeed these registers are read directly from the inverter with no calculations involved.

I wonder whether there's been a firmware update recently which broke them...

zaapp, 2023-07-21:
It is likely that Chint DTSU666 is to blame for this situation. After switching off the mains power the unit rebooted and now the energy reading is correct.

canton7, 2023-07-22:
Aha good to know thanks! I'll close this off. If you see a repeat, please add to #362.

closed 2023-07-22 by canton7