nathanmarlor/foxess_modbus #374: Charge Current Setting

FozzieUK, 2023-07-23:
Hi Antony,

I have a custom charge profile to manage the battery charge as it nears full - until yesterday I used my own script which wrote to the max dc charge register (41007) but as you have added this as a number to the integration recently I thought I would simplify things, drop my script and update **number.max_charge_current** via a service call.

When I checked logs this morning there is an occasional inconsistency in the charge current I ask for, and the one I get - I use an entity rounded to 1 decimal place (and step size of 0.1) to set it, I have double checked it's value when I set the **number.max_charge_current** and it is correct - weirdly and just occasionally I get a 0.1A difference as if there is a float conversion error somewhere.

For example from my logs my charge profile sets **number.max_charge_current** -
set 10.0A, and get 10.0A
set 8.0 get 8.0
set 4.9 get 4.9
set 4.2 get 4.2
set 3.6 get 3.6
set 3 get 3
set 2 get 2

But when I set -
set 2.8 get 2.7
set 6.6 get 6.5
set 2.4 get 2.3

If I use the HA card and set the entity **number.max_charge_current** I get the right answer it's only when I set it through a service call that it doesn't work and only on some values.

my entity **input_number.battery_charge_limit** has step size of 0.1 and I round it to 1 dp when I set it, and again when I set the service call - I tried changing the round(1) to a float but it makes no difference.

The service call is

```
service: number.set_value
data:
  value: "{{ states('input_number.battery_charge_limit')|round(1,0) }}"
target:
  entity_id: number.max_charge_current
```

I even tried this simplified version and get the same result-

[image]

This is my logbook just now when I try and set 2.8
  
[image]

It's obviously not a big issue and doesn't really affect my charge profile I thought i'd mention it just in case there are any undesired side effects !

canton7, 2023-07-23:
Thanks!

Replicated, and it's a rounding error in my code. Fix incoming...

closed 2023-07-23 by canton7

FozzieUK, 2023-07-23:
Great, thank you 👍

PR #375 (2023-07-23): Round rather than truncating number values
Floating-point inaccuracy means that e.g. 6.6 * 10 = 65.999999.

Fixes: #374
commit 8f567075 (2023-07-23): Round rather than truncating number values

Floating-point inaccuracy means that e.g. 6.6 * 10 = 65.999999.

Fixes: #374