nathanmarlor/foxess_modbus #414: Please remove requirement that end time is before start time - Fox have fixed/removed this

canton7, 2023-09-12:
### Discussed in https://github.com/nathanmarlor/foxess_modbus/discussions/413

<div type='discussions-op-text'>

<sup>Originally posted by **iancg** September 12, 2023</sup>
Hi,

Please see:

https://www.facebook.com/groups/foxessownersgroup/posts/1923430298039143/?comment_id=1923434278038745&reply_comment_id=1924573837924789

I've tested this and with software Master 1.60/Manager 1.63, using start=23:30 and end=05:30 worked as expected.

Can the check that prevents start > end be removed please.

Setting on web UI:

[image]

Battery charged overnight (it was full just before 3am, but started at 11:30 and continued over midnight and didn't start discharging until 5:30am).

[image]
</div>

lamixer, 2023-09-27:
I'd like to figure out how to set this in the HA integration at all! I can't spot what entity it is.

canton7, 2023-09-27:
There's a service you can call, or you can use the charge period card linked from the readme.

lamixer, 2023-09-29:
Thanks canton7 -- I missed that. I'm trying to get the card to work, isn't quite yet though! I started a discussion.

PR #428 (2023-09-30): Remove restictions on overlapping charge periods
Foxcloud has removed all restrictions. Mirror this.

Fixes: #414
commit abe0bf95 (2023-09-30): Remove restictions on overlapping charge periods

Foxcloud has removed all restrictions. Mirror this.

Fixes: #414
closed 2023-09-30 by canton7