nathanmarlor/foxess_modbus #422: Phases on H3 can be < 0

canton7, 2023-09-22:
Ahhh right, now I understand the problem you've got, - yes you are correct there is a range check on the inv_power_r/s/t and inv_current_r/s/t that doesn't allow these values to be negative.

It'll need @canton7 to have a look at that, but for now you could test it manually by changing the range checks (around lines 621/633/645 of /config/custom_components/foxess_modbus/entities/entity_descriptions.py and make the range check 'validate=[Range(-100, 100)]' and then restart your home assistant.

_Originally posted by @FozzieUK in https://github.com/nathanmarlor/foxess_modbus/discussions/362#discussioncomment-7083283_

TonyM1958, 2023-09-22:
Not sure I've fully understood the scenario here - but also check if you have a problem with the consolidated power sensors that sum R, S and T and present the data as a virtual single phase inverter.

I previously separated +ve and -ve RST values and summed the +ve and -ve values separately but found that this resulted in a non-zero grid consumption when the inverter was balancing the load. The inverter will happily import on 1 phase and export on another. As long R + S + T = 0, the inverter considers the load to be balanced.

If you have slaves - it would be interesting to understand how the RST power values on the master behave in relation to load power - does the master present just it's own RST power and load or sum the RST power across the master and slaves? It's analogous to the way CT2 works when a solar inverter connected - execpt you don't have CT2 or meter2 getting the power from the slave inverters...

canton7, 2023-09-30:
@TonyM1958 Probably best to reply in the discussion -- noone there is notified of comments here.

PR #429 (2023-10-04): Remove restrictions on inverter power and EPS power being >=0 for H3
Seems that negative values are expected and part of normal operation.

Fixes: #422
commit 9f622188 (2023-10-04): Remove restrictions on inverter power and EPS power being >=0 for H3

Seems that negative values are expected and part of normal operation.

Fixes: #422
closed 2023-10-04 by canton7