nathanmarlor/foxess_modbus #430: "Unable to prepare setup for platform foxess_modbus.binary_sensor: Platform not found (No module named 'custom_components.foxess_modbus.binary_sensor')."

calum-mcfarlane, 2023-10-06:
## 1.74

## Configuration

```yaml
Logger: homeassistant.setup
Source: setup.py:359
First occurred: 09:07:14 (1 occurrences)
Last logged: 09:07:14

Unable to prepare setup for platform foxess_modbus.binary_sensor: Platform not found (No module named 'custom_components.foxess_modbus.binary_sensor').
```

## Describe the bug

Yesterday I tried to upgrade to the latest version of the integration and custom card via HACS. The card upgrade went fine, but the integration upgrade got 'stuck' (progress wheel kept on going and going) - all sensors for the inverter became "unavailable". I tried to re-download the previous version, which got most of the sensors working, but the status flags for both charge periods remain unavailable. In addition, I see the error message posted above in the logs.

I also see that the charge period update service is no longer working:

[image]
I only have one inverter so I haven't ever given mine a "friendly name" when setting up the integration

## Debug log

```text

If I enable debug logging for the integration I just see lots of sensors being read but no further errors - it only seems to throw that error when trying to set up, but after that's gone wrong, other things seem to fail silently

```

FozzieUK, 2023-10-06:
Not sure how much help this is, but I upgraded to the new version ok - with no errors ( I was on HA 2023.9.x but now moved on  to 2023.10.0)

From the error and your description of it getting stuck it sounds as if it failed to update some of the base image (files locked?), I would be tempted to completely remove the integration from Devices & Services, then remove it from HACS - clean restart HA, and re-add it again via HACS.

calum-mcfarlane, 2023-10-06:
Thanks Dave. This is turning into a bit of a headscratcher. So far, I've 

- removed the integration from D&S and then HACS ,as suggested
- restarted HA
- then removed the custom card for changing charge periods
- then restarted HA again
- then re-downloaded the integration from HACS, re-added and reconfigured it
- confirmed all the sensors that should be available, are available (all looking good, the charge start/end and status flag sensors were unavailable before but are now showing as working again)
- I then re-downloaded the custom card
- re-added it to the relevant dashboard

This is where it gets a bit weird - I have an ordinary card showing me the status of the charge period entities, it looks like this:

[image]
This is as expected and in line with what I can see in the Foxcloud website.

However the service to update the change periods still gives the same error as my original report, and if I tried to change the MaxSOC parameter I get this error:

```
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:226
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 10:59:31 (1 occurrences)
Last logged: 10:59:31

[139865491503424] Error writing registers. Start: 41010; values: [96]; slave: 247 from tcp://192.168.0.41:502: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 226, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 2012, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2049, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 876, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 948, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 105, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/foxess_modbus/entities/modbus_number.py", line 107, in async_set_native_value
    await self._controller.write_register(self._address, int_value)
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 96, in write_register
    await self.write_registers(address, [value])
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 121, in write_registers
    raise ex
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 108, in write_registers
    await self._client.write_registers(start_address, values, self._slave)
  File "/config/custom_components/foxess_modbus/modbus_client.py", line 272, in write_registers
    raise ModbusClientFailedError(message, self, response) from response
custom_components.foxess_modbus.modbus_client.ModbusClientFailedError: Error writing registers. Start: 41010; values: [96]; slave: 247 from tcp://192.168.0.41:502: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)

```

The charge period card looks like this: 
[image]

At this point I was wondering if this is a problem with my (hitherto very reliable) 410S or some weird snafu with HA. I rebooted the 410S and the ability to change the MaxSOC from a lovelace card started working again (ie I can change it on the control, and a couple of seconds later the sensor changes). However the charge period card remains broken, as does the service which it calls. Throughout all of this, all the normal sensors are being read out without a problem from the inverter. 

It feels like the attempted adding / re-downloading etc of the integration has somehow got things mixed up, and now it's looking for a friendly name for the inverter despite there never having been one in the past. I'm reluctant to re-configure the integration with a name in case it creates a whole slew of new sensors, but I'm not sure where to go from here!

FozzieUK, 2023-10-06:
That is a strange one, never seen that before.

If the 410S itself was causing MaxSoC not to be changed (basically blocking register writes), and after a reboot it is ok - I think if you were to restart HA again (which basically reloads the integration and re-creates the missing sensors) - then it should work, fingers crossed ;)

calum-mcfarlane, 2023-10-06:
So I did do "Hail Mary" HA restart, "just in case" and the Charge Period card is now working again. I can make changes on the card and see them reflected in the sensors almost immediately. I also realised that the Service was **probably** always working, I had neglected to choose the FoxESS Modbus entry from the pulldown 😵

So, everything's working again, but I wonder, is there any interaction between the custom card and the integration that could have caused this? Should I update them in a specific way to avoid such problems in future? The "cleanest" way would seem to be to restart HA after updating each and every item in HACS, which is tedious but better than all this faffing about 😄

FozzieUK, 2023-10-06:
It's possible, a newer version of card with the safety removed around the charge time setting interfacing the old integration, but none of it would persist after a reboot so it must have messed up the 410S in some way :confused: - probably one for Antony to think about that 👍 
I upgraded the integration first, then the card and that all worked fine - but i'll keep that in mind when I update a friends W610... they're enough trouble already 😉

canton7, 2023-10-06:
The only interactions between the card and the integration are:

1. The card asks the integration for the names of the charge-related sensor
2. The integration sets those sensors as normal; the card reads them
3. The card calls the "Set charge periods" service to set the charge periods

The latest card only removes a validation check about overlapping charge periods. None of the above looks in any way related to the card. 

The main problem I think is that `custom_components/foxess_modbus/binary_sensor.py` was missing after the upgrade. Not sure why. That's indicated by "No module named 'custom_components.foxess_modbus.binary_sensor'" error.

That probably caused the integration to half set up. Most of the sensors were set up, but it never got to the point where it added the inverter to the list of inverters it keeps internally. That explains the card and service not being able to find the inverter.

A secondary problem is that the 410S stopped responding. That's what "Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)" indicates. This will caused foxess_modbus to show the sensors as normal until we get a certain number of timeouts, then it makes them unavailable -- this can take around a minute. So when you looked at your ordinary card it probably hadn't timed out yet, but when you looked at the charge period card the sensors had become unavailable. This also explains why the service writes failed.

calum-mcfarlane, 2023-10-06:
Thanks both. I'll mark this one as resolved. Hopefully the 410S doesn't play up any more.

closed 2023-10-06 by calum-mcfarlane