nathanmarlor/foxess_modbus #457: Sensors stop working

djdarlo, 2023-11-03:
## Version of the custom_component

## Configuration

```yaml
Add your logs here.
```

## Describe the bug

A clear and concise description of what the bug is.

I am using the modbus serial integration. Every so often the sensors stop reporting. A simple restart of HA sorts it but I have noticed these errors in the log. 

## Debug log

```text

Add your logs here.

This error originated from a custom integration.

Logger: custom_components.foxess_modbus.modbus_controller
Source: custom_components/foxess_modbus/modbus_controller.py:220
Integration: FoxESS - Modbus (documentation, issues)
First occurred: 06:35:45 (1 occurrences)
Last logged: 06:35:45

/dev/ttyUSB0 247 - 5 failed poll attempts: now not connected. Last error: filedescriptor out of range in select()

This error originated from a custom integration.

Logger: custom_components.foxess_modbus.modbus_controller
Source: custom_components/foxess_modbus/modbus_controller.py:152
Integration: FoxESS - Modbus (documentation, issues)
First occurred: 06:34:57 (800 occurrences)
Last logged: 09:54:46

General exception when polling /dev/ttyUSB0 247: ValueError('filedescriptor out of range in select()')
Traceback (most recent call last):
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 152, in _refresh
    reads = await self._client.read_registers(
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/foxess_modbus/modbus_client.py", line 210, in read_registers
    response = await self._async_pymodbus_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/foxess_modbus/modbus_client.py", line 298, in _async_pymodbus_call
    result = await self._hass.async_add_executor_job(call, *args)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/mixin.py", line 116, in read_input_registers
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 192, in execute
    return self.transaction.execute(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 159, in execute
    response, last_exception = self._transact(
                               ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 275, in _transact
    size = self._send(packet)
           ^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 323, in _send
    return self.client.framer.sendPacket(packet)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/framer/rtu_framer.py", line 296, in sendPacket
    size = self.client.send(message)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/serial.py", line 278, in send
    size = self.socket.write(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/serial/serialposix.py", line 640, in write
    abort, ready, _ = select.select([self.pipe_abort_write_r], [self.fd], [], None)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: filedescriptor out of range in select()

```

canton7, 2023-11-03:
Looks the same as #273, but for USB devices rather than network devices. 

The problem is that you have too many file descriptors open (more than 1024), and Python's serial library can't cope.

I'm not honestly sure what we're going to be able to do about it, as the problematic code is two layers of abstraction below us.

canton7, 2023-11-03:
Hmm, looks like pyserial has a `PosixPollSerial` class, which will do what we want. Trying to work out how to get pymodbus to use it...

canton7, 2023-11-03:
https://pyserial.readthedocs.io/en/latest/url_handlers.html#alt. I'll try and give that a try over the weekend, although I'll need some help testing as I don't have a serial USB device to test with.

djdarlo, 2023-11-03:
Happy to give it a test for you. I’ve got a couple of HA setups using this.

canton7, 2023-11-04:
Reopening until we've confirmed this works

canton7, 2023-11-04:
@djdarlo Would you mind trying out the pre-release v1.8.0b3, see if that fixes it for you? Thanks!

djdarlo, 2023-11-04:
Of course. Will try it later today.

djdarlo, 2023-11-04:
How do I download a pre release?

canton7, 2023-11-04:
Hacs - Integrations - FoxESS Modbus - Hamburger menu in the top right - Redownload, then select the "Show beta versions" toggle

djdarlo, 2023-11-04:
Thank you. I’ve upgraded one and left one as is. Working. I’ll keep you
updated on how it goes. Thanks again.

djdarlo, 2023-11-06:
The other one errored yesterday so I've updated that one too. Both going fine so far. I'll keep you updated. Thank you.

djdarlo, 2023-11-07:
I spoke too soon.  One of my HA installs went pop over night :( Other one is still going fine.  Logs below...

1)
This error originated from a custom integration.

Logger: custom_components.foxess_modbus.modbus_controller
Source: custom_components/foxess_modbus/modbus_controller.py:129
Integration: FoxESS - Modbus (documentation, issues)
First occurred: November 5, 2023 at 8:30:16 PM (1 occurrences)
Last logged: November 5, 2023 at 8:30:16 PM

Aborting refresh of /dev/ttyUSB0 247 as a previous refresh is still in progress. Is your poll rate '15' too high?

2)
This error originated from a custom integration.

Logger: custom_components.foxess_modbus.modbus_controller
Source: custom_components/foxess_modbus/modbus_controller.py:153
Integration: FoxESS - Modbus (documentation, issues)
First occurred: 1:45:24 AM (3 occurrences)
Last logged: 1:47:24 AM

General exception when polling /dev/ttyUSB0 247: UnboundLocalError("cannot access local variable 'fd' where it is not associated with a value")
Traceback (most recent call last):
  File "/config/custom_components/foxess_modbus/modbus_controller.py", line 153, in _refresh
    reads = await self._client.read_registers(
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/foxess_modbus/modbus_client.py", line 218, in read_registers
    response = await self._async_pymodbus_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/foxess_modbus/modbus_client.py", line 306, in _async_pymodbus_call
    result = await self._hass.async_add_executor_job(call, *args)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/mixin.py", line 117, in read_input_registers
    pdu_reg_read.ReadInputRegistersRequest(address, count, slave, **kwargs)
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 177, in execute
    self.delay_ms = self.params.reconnect_delay
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 168, in execute
    )
      
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 313, in _transact
    ) as msg:
              
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 392, in _recv
    result = read_min + result
             ~~~~~~~~~^~~~~~~~~
  File "/usr/local/lib/python3.11/site-packages/pymodbus/framer/rtu_framer.py", line 296, in recvPacket
    size = self.client.send(message)
             ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/serial.py", line 271, in recv
    return result
  File "/usr/local/lib/python3.11/site-packages/serial/serialposix.py", line 830, in read
    if fd == self.pipe_abort_read_r:
       ^^
UnboundLocalError: cannot access local variable 'fd' where it is not associated with a value

3)
This error originated from a custom integration.

Logger: custom_components.foxess_modbus.modbus_controller
Source: custom_components/foxess_modbus/modbus_controller.py:221
Integration: FoxESS - Modbus (documentation, issues)
First occurred: 1:50:23 AM (1 occurrences)
Last logged: 1:50:23 AM

[... 37 lines cut ...]
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/mixin.py", line 117, in read_input_registers
    pdu_reg_read.ReadInputRegistersRequest(address, count, slave, **kwargs)
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 177, in execute
    self.delay_ms = self.params.reconnect_delay
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 168, in execute
    )
      
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 285, in _transact
    return size, None
       ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 325, in _send
    def _recv(self, expected_response_length, full):  # pylint: disable=too-complex
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/framer/rtu_framer.py", line 286, in sendPacket
    break
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/serial.py", line 231, in send
    self.last_frame_end = None
       ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/serial/serialposix.py", line 640, in write
    abort, ready, _ = select.select([self.pipe_abort_write_r], [self.fd], [], None)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: filedescriptor out of range in select()
[home-assistant_2023-11-07T06-48-03.419Z.zip](https://github.com/nathanmarlor/foxess_modbus/files/13276614/home-assistant_2023-11-07T06-48-03.419Z.zip)

canton7, 2023-11-07:
Are you sure you're running v1.8.0b3? The line numbers in that stack trace line up with v1.7.6, but not v1.8.0b3.

djdarlo, 2023-11-07:
I think so....

[image: image.png]

canton7, 2023-11-07:
You need to go to the github web interface to upload images

canton7, 2023-11-07:
Aha, no, that's my mistake.

Slightly confused by what's going on in pyserial there. I'll have a deeper look when I get the chance

canton7, 2023-11-29:
Sorry for the delay!

Looks like this is a known issue in pyserial: https://github.com/pyserial/pyserial/issues/617.

I've had a go at working around the issue. Can you give v1.8.0b5 a try please?

djdarlo, 2023-11-29:
> Looks like this is a known issue in pyserial: pyserial/pyserial#617

Of course. Will put it one of my systems now. I’d currently got a daily
restart in place to work around it so will remove that too.

D

canton7, 2023-12-09:
Hi, any news on whether it's behaving better?

djdarlo, 2023-12-09:
Hey there. It seemed to fail again so I have reverted to a daily restart.
I’ll capture the logs next time and send them over to you.

canton7, 2023-12-09:
Thanks! It's likely a different problem to last time. I'm afraid I don't have a serial devie here, so I'm having to develop this workaround by proxy!

djdarlo, 2023-12-11:
Not a problem - its just gone pop again - here you go :)

Home Assistant Core

*Connection to (192.168.1.102, 502) failed: [Errno 113] Host is
unreachable3:32:36 PM – (ERROR) components/modbus/modbus.py - message first
occurred at December 10, 2023 at 11:19:32 PM and shows up 37187
timesGeneral exception when polling /dev/ttyUSB0 247:
ValueError('filedescriptor out of range in select()')3:32:34
PM – (WARNING) FoxESS - Modbus (custom integration) - message first
occurred at 3:30:21 PM and shows up 14 timesCleanup recv buffer before
send: 0xf7 0x4 0xba 0xc 0x81 0x0 0x1 0x0 0x24 0x5 0x5f 0x0 0x0 0x0 0x3 0x9
0x5d 0x0 0x13 0x1 0xd0 0x9 0x5d 0x0 0x15 0x1 0xd9 0xff 0xdf 0x1 0xfc 0x13
0x7f 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xff 0xf8 0x0 0x0 0x1
0xe6 0x1 0x52 0x1 0xb1 0x0 0x0 0x0 0x0 0x17 0x70 0xe8 0x90 0x0 0x0 0x0 0x0
0x2 0x2 0x5 0x5 0x9 0x5a 0xff 0xee 0x0 0x4e 0x3 0x8a 0x1 0xe 0xa 0x44 0x7
0x8 0x1 0xf4 0x1 0xf4 0x1 0x1f 0x0 0xe3 0xc 0xf7 0xc 0xeb 0x0 0x4 0x0 0xd1
0x79 0x95 0x0 0xa 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x0 0x0 0x0 0x2 0x0 0x1
0x0 0x1 0x0 0x1 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
0x0 0x0 0x0 0x0 0x0 0xd1 0xde 0x0 0x2e 0x0 0x0 0x5d 0x49 0x0 0x64 0x0 0x0
0x56 0xf2 0x0 0x31 0x0 0x0 0x4d 0x8d 0x0 0x1 0x0 0x0 0x44 0x16 0x0 0x5a 0x0
0x0 0xde 0x2e 0x0 0x48 0x0 0x0 0x28 0x49 0x0 0x58 0x0 0x0 0xaa 0x42 0x0
0x4a 0x81 0xaf3:32:34
PM – (WARNING) /usr/local/lib/python3.11/site-packages/pymodbus/logging.py
- message
first occurred at 2:30:23 AM and shows up 17 times/dev/ttyUSB0 247 - 5
failed poll attempts: now not connected. Last error: filedescriptor out of
range in select()3:30:54 PM – (WARNING) FoxESS - Modbus (custom
integration)Error writing holding register - retry (1/5): Modbus Error:
[Input/Output] No Response received from the remote slave/Unable to decode
response3:30:18 PM – (WARNING) FoxESS - Energy Management (custom
integration) - message first occurred at 2:30:29 AM and shows up 4
timesAborting refresh of /dev/ttyUSB0 247 as a previous refresh is still in
progress. Is your poll rate '10' too high?11:30:13 AM – (WARNING) FoxESS -
Modbus (custom integration) - message first occurred at December 10, 2023
at 11:28:04 PM and shows up 2 timesSomething is blocking Home Assistant
from wrapping up the start up phase. We're going to continue anyway. Please
report the following info at https://github.com/home-assistant/core/issues
<https://github.com/home-assistant/core/issues>: humidifier, input_button,
fan.tuya, camera, application_credentials, text, sensor.integration,
hardware, notify.mobile_app, event.octopus_energy, solcast_solar,
time.octopus_energy, ring, influxdb, light.tapo, climate.meross_cloud,
assist_pipeline, websocket_api, sensor.template, siren.tuya,
binary_sensor.rpi_power, bluetooth, persistent_notification,
light.smartthings, binary_sensor.ring, binary_sensor, input_boolean,
vacuum.tuya, pirateweather, sensor.statistics, file_upload,
sensor.meross_cloud, repairs, radio_browser, vacuum, calendar,
sensor.co2signal, sensor.oralb, co2signal, sensor.inkbird,
media_player.dlna_dmr, search, switch.tuya, ssdp, sensor.foxess, rpi_power,
climate.hive, sensor.hue, alarm_control_panel.tuya, switch.harmony,
sensor.esphome, hive, light.tuya, light.hue, bluetooth_adapters,
system_log, webhook, light.hive, stt, device_tracker.mobile_app,
image_upload, light.meross_cloud, select, sensor.derivative, cover.tuya,
binary_sensor.modbus, button.tuya, http, select.foxess_modbus, api,
siren.ring, speedtestdotnet, sensor.ring, switch.meross_cloud,
switch.foxess_em, person, number.foxess_em, binary_sensor.hue, integration,
switchbot, device_automation, switch.switchbot, sensor.speedtestdotnet,
binary_sensor.hassio, lock.smartthings, sensor.tapo, sensor.modbus, siren,
meross_cloud, cover.smartthings, binary_sensor.hive, alarm_control_panel,
fan, camera.ring, update.hassio, onboarding, counter, stream, logger,
tts.google_translate, humidifier.tuya, energy, input_text, sensor.moon,
recorder, light.ring, water_heater, sun, tapo, schedule, foxess_modbus,
homeassistant, raspberry_pi, ipp, climate.tuya, octopus_energy, scene.hue,
sensor.solcast_solar, binary_sensor.mobile_app,
binary_sensor.octopus_energy, scene, switch.ring, number.tuya,
sensor.hassio, sensor.foxess_modbus, sensor.octopus_energy, hassio, usb,
conversation, sensor.aurora, climate.smartthings, select.tuya,
input_number, zeroconf, binary_sensor.aurora, my, fan.smartthings,
system_health, remote.harmony, input_datetime, weather, default_config,
timer, sensor.ibeacon, aurora, script, climate, diagnostics,
[... 127 lines cut ...]
0x0 0x58 0x0 0x0 0xaa 0x42 0x0 0x4a 0xc8 0x2d

This error originated from a custom integration.

Logger: custom_components.foxess_modbus.modbus_controller
Source: custom_components/foxess_modbus/modbus_controller.py:221
Integration: FoxESS - Modbus (documentation, issues)
First occurred: 3:30:54 PM (1 occurrences)
Last logged: 3:30:54 PM

/dev/ttyUSB0 247 - 5 failed poll attempts: now not connected. Last error:
filedescriptor out of range in select()

This error originated from a custom integration.

Logger: custom_components.foxess_em.fox.fox_modbus
Source: custom_components/foxess_em/fox/fox_modbus.py:89
Integration: FoxESS - Energy Management (documentation, issues)
First occurred: 2:30:29 AM (4 occurrences)
Last logged: 3:30:18 PM

Error writing holding register - retry (1/5): Modbus Error: [Input/Output]
No Response received from the remote slave/Unable to decode response

D

canton7, 2023-12-11:
Thanks! Are you sure you're running 1.8.0b5? We shouldn't be using serialposix.py any more.

That said, my replacement for serialposix.py might still have an issue on that line - I'll look into it.

djdarlo, 2023-12-11:
Yep. Just checked. 1.8.0b5.

canton7, 2023-12-14:
Right! I've had a chance to properly look at this, and it turns out that pyserial's attempt to use `poll` rather than `select` just addresses the read operation, and not the write operation.

So I've updated my workaround to also provide a write implementation which uses `poll`. Can you give v1.8.0b6 a try please?

djdarlo, 2023-12-14:
Of course

djdarlo, 2023-12-14:
Done and disabled my nightly restart. Will report back.

D

canton7, 2023-12-17:
Hi, any news?

djdarlo, 2023-12-17:
All good so far :)

canton7, 2023-12-17:
Good to hear, thanks!

canton7, 2023-12-23:
Hi, is it still holding up a week in?

djdarlo, 2023-12-23:
Yep. All good. Think you’ve cracked it :)

canton7, 2023-12-23:
Fab, thank you! I'll get that released properly at some point.

PR #461 (2023-11-04): Use PosixPollSerial from pyserial for serial connections where possible
This uses poll rather than select, which means we're not limited to 1024 fds.

Fixes: #457
commit bef5065b (2023-11-04): Use PosixPollSerial from pyserial for serial connections where possible

This uses poll rather than select, which means we're not limited to 1024 fds.

Fixes: #457
closed 2023-12-23 by canton7