nathanmarlor/foxess_modbus #463: Allow setting time / date with a service

canton7, 2023-11-05:
On a 5 month period, I would say the internal clock is about 8 min behind.
I disconnected my inverter from the internet the end of May. I reconnected it today in the hope DST would be applied, but alas ... I had to set it manually and that's when I saw the time was 52 minutes ahead when it should be 1:00 :)

So @canton7, any news on that time/date integration? 😇
I would love to be able to read the internal time of the inverter and set it via HA.

_Originally posted by @screenagerbe in https://github.com/nathanmarlor/foxess_modbus/discussions/293#discussioncomment-7479676_

WyndStryke, 2025-02-08:
Don't know if these fields are writeable, but I can see that the time is drifting further (8 February 2025 at 23:12:32 versus inverter 25/2/8 23:14:27), about 4 seconds per day.  The inverter is online, has recent firmware, KH7.  Do they normally time sync?

```
id: '1737718182030'
alias: RTCWriteDateTimeToLog
description: Write date & time from inverter into HA log
triggers:
  - trigger: time
    at: '00:00:00'
conditions: []
actions:
  - variables:
      RTCRegister: null
  - action: foxess_modbus.read_registers
    data:
      start_address: 40000
      type: input
      inverter: eebf190027c514026eba9631f55d1760
      count: 6
    response_variable: RTCRegister
  - action: logbook.log
    metadata: {}
    data:
      name: RTCStoreInLog
      message: >-
        {% set val = RTCRegister['error'] %} {% if 'Error reading registers' not
        in val %}
          {% set year = RTCRegister['values'].40000|string %}
          {% set month = RTCRegister['values'].40001|string %}
          {% set day = RTCRegister['values'].40002|string %}
          {% set hour = RTCRegister['values'].40003|string %}
          {% set minute = RTCRegister['values'].40004|string %}
          {% set second = RTCRegister['values'].40005|string %}
          RTC Date Time received {{year}}/{{month}}/{{day}} {{hour}}:{{minute}}:{{second}}
        {% else %}
          No RTC Time, received {{ RTCRegister['error'] }}
        {% endif %}
mode: single
```

```
Executed: 8 February 2025 at 23:12:32
Result:
params:
  domain: logbook
  service: log
  service_data:
    name: RTCStoreInLog
    message: RTC Date Time received 25/2/8 23:14:27
  target: {}
running_script: false
```

FozzieUK, 2025-02-09:
The Fox servers used to sync the inverter clocks regularly but that appears to have stopped after last years clock change - we've reported it to them and they have now acknowledged that it has been broken and they are working on a fix but for now you should check your inverter time manually. 

My H1 loses 1 second each day fairly reliably, whereas I believe the KH is a bit more generous in it's time loss - when it worked (and it will hopefully again soon) my clock was always bang on.

You can write to the clock registers, they have to be written as a span of 6 and despite the fact the read on the year register returns 2025 you can only write the '25' to it, anything else fails - for example the string to write to 40000 (for 6) to set the clock would be 25,02,09,08,50,42

WyndStryke, 2025-02-09:
The following does not work, so don't try using it, but I'm not sure why it doesn't work (very much a HA/yaml newbie).

It is dropping out at the 'safe_to_update' check near the end, so I'm guessing that this might be some sort of variable scope error, but if they're defined on the top I'm not sure why they're not getting set?  

Also I'm not sure if I am writing the registers right, passing in integers, will that generate a CSV? (not that the code has reached that point yet), or should I be converting them to strings as I go?  TBH this is making me quite nervous.  I don't want to accidentally write junk into the RTC.

```
            data:
              start_address: 40000
              values: >-
                {{(new_year),(new_month),(new_day),(new_hour),(new_minute),(new_second)}}

```

```
alias: RTC log inverter time and update
description: Write date & time from inverter into HA log
triggers:
  - trigger: time
    at: "00:00:00"
conditions: []
actions:
  - variables:
      RTCRegister: null
      year: null
      month: null
      day: null
      hour: null
      minute: null
      second: null
      new_year: null
      new_month: null
      new_day: null
      new_hour: null
      new_minute: null
      new_second: null
      safe_to_update: false
      need_to_update: false
  - action: foxess_modbus.read_registers
    data:
      start_address: 40000
      type: input
      inverter: eebf190027c514026eba9631f55d1760
      count: 6
    response_variable: RTCRegister
  - action: logbook.log
    metadata: {}
    data:
      name: RTCStoreInLog
      message: >-
        {% set val = RTCRegister['error'] %} {% if 'Error reading registers' not
        in val %}
          {% set year = RTCRegister['values'].40000| int %}
          {% set month = RTCRegister['values'].40001| int %}
          {% set day = RTCRegister['values'].40002| int %}
          {% set hour = RTCRegister['values'].40003| int %}
          {% set minute = RTCRegister['values'].40004| int %}
          {% set second = RTCRegister['values'].40005| int %}
          {% set new_year = now().strftime('%y') | int   %}
          {% set new_month = now().strftime('%m') | int  %}
          {% set new_day = now().strftime('%d') | int  %}
          {% set new_hour = now().strftime('%H') | int  %}
          {% set new_minute = now().strftime('%M') | int  %}
          {% set new_second = now().strftime('%S') | int  %}      
          RTC Date Time received {{year}}/{{month}}/{{day}} {{hour}}:{{minute}}:{{second}}
          HA Date Time {{new_year}}/{{new_month}}/{{new_day}} {{new_hour}}:{{new_minute}}:{{new_second}}
          {% if year == new_year and month == new_month and hour == new_hour and minute == new_minute and second == new_second %}
            Inverter RTC time matches HA time
            {% set need_to_update = false %}
          {% else %}
            Inverter RTC time does not match HA time
            {% set need_to_update = true %}
          {% endif %}
          {% if year == new_year and month == new_month and hour == new_hour and (((minute-new_minute)|abs) < 5) %}
            Safe to update (no change in date or hour, minutes differ by less than 5)
            {% set safe_to_update = true %}
          {% else %}
            Unsafe to update (difference too large)
            {% set safe_to_update = false %}
          {% endif %}
        {% else %}
          No RTC Time, received {{ RTCRegister['error'] }}
        {% endif %}
  - choose:
      - conditions:
          - condition: template
            value_template: "{{ not (safe_to_update | bool) }}"
        sequence:
          - action: logbook.log
            metadata: {}
            data:
              name: RTCStoreInLog
              message: Unsafe to update
      - conditions:
          - condition: template
            value_template: "{{ not (need_to_update | bool) }}"
        sequence:
          - action: logbook.log
            metadata: {}
            data:
              name: RTCStoreInLog
              message: No need to update
      - conditions:
          - condition: template
            value_template: >-
              {{ ((year <= 0) or (new_month < 1)  or (new_month > 12) or
              (new_day < 1) or (new_day > 31) or (new_hour < 0) or (new_hour >
              23) or (new_minute < 0) or (new_minute > 59) or (new_second < 0) 
              or (new_second > 59)) }}
        sequence:
          - action: logbook.log
            metadata: {}
            data:
              name: RTCStoreInLog
              message: New time invalid
      - conditions:
          - condition: template
            value_template: >-
              {{ (safe_to_update | bool) and (need_to_update | bool) and ((year
              <= 0) or (new_month < 1)  or (new_month > 12) or (new_day < 1) or
              (new_day > 31) or (new_hour < 0) or (new_hour > 23) or (new_minute
              < 0) or (new_minute > 59) or (new_second < 0)  or (new_second >
              59)) }}
        sequence:
          - action: foxess_modbus.write_registers
            metadata: {}
            data:
              start_address: 40000
              values: >-
                {{(new_year),(new_month),(new_day),(new_hour),(new_minute),(new_second)}}
              inverter: eebf190027c514026eba9631f55d1760
          - action: logbook.log
            metadata: {}
            data:
              name: RTCStoreInLog
              message: Updated time
mode: single
```

FozzieUK, 2025-02-09:
The main problem is 'variables' are of local scope, so if you set them in an action they will only be available at that level, in the next action they will be back to null - hence your need_to_update, safe_to_update bools won't hold a value once that action is complete. 

However the RTCRegister['values'] .xxxxx will be available throughout so you can re-generate the variables at each level (or use something like a helper input number and set it once then test the helper through the automation)

There's also an error in this line `{% if year == new_year and month == new_month and hour == new_hour and (((minute-new_minute)|abs) < 5) %}` as the year received from the inverter registers is returned as 2025 whereas %y returns 25 - it's **only** the write to the inverter year register where you have to send 25 (not 2025)

WyndStryke, 2025-02-09:
Many thanks, I think I'm a lot closer now - one more issue I also don't understand.  It runs all the way through, but seems to fall over in write_registers.  It's complaining about an open bracket in the year literal, but the generated parameters look OK to me?

```
Executed: 9 February 2025 at 17:41:22
Error: invalid literal for int() with base 10: '(25'
Result:
params:
  domain: foxess_modbus
  service: write_registers
  service_data:
    start_address: 40000
    values:
      - 25
      - 2
      - 9
      - 17
      - 41
      - 22
    inverter: eebf190027c514026eba9631f55d1760
  target: {}
running_script: false
```

```
          - action: foxess_modbus.write_registers
            metadata: {}
            data:
              start_address: 40000
              values: >-
                {{now().strftime('%y')|int,now().strftime('%m')|int,now().strftime('%d')|int,now().strftime('%H')|int,now().strftime('%M')|int,now().strftime('%S')|int}}
              inverter: eebf190027c514026eba9631f55d1760
```

```
alias: RTC log inverter time and update
description: Write date & time from inverter into HA log
triggers:
  - trigger: time
    at: "00:00:00"
conditions: []
actions:
  - variables:
      RTCRegister: null
      year: null
      month: null
      day: null
      hour: null
      minute: null
      second: null
      new_year: null
      new_month: null
      new_day: null
      new_hour: null
      new_minute: null
      new_second: null
      safe_to_update: false
      need_to_update: false
  - action: foxess_modbus.read_registers
    data:
      start_address: 40000
      type: input
      inverter: eebf190027c514026eba9631f55d1760
      count: 6
    response_variable: RTCRegister
  - action: logbook.log
    metadata: {}
    data:
      name: RTCStoreInLog
      message: >-
        {% set val = RTCRegister['error'] %} {% if 'Error reading registers' not
[... 119 lines cut ...]
              RTCRegister['values'].40005| int %} {% set new_year =
              now().strftime('%y') | int   %} {% set new_month =
              now().strftime('%m') | int  %} {% set new_day =
              now().strftime('%d') | int  %} {% set new_hour =
              now().strftime('%H') | int  %} {% set new_minute =
              now().strftime('%M') | int  %} {% set new_second =
              now().strftime('%S') | int  %}             {{ not ((year < 25) or
              (new_month < 1)  or (new_month > 12) or (new_day < 1) or (new_day
              > 31) or (new_hour < 0) or (new_hour > 23) or (new_minute < 0) or
              (new_minute > 59) or (new_second < 0)  or (new_second > 59)) }}
        sequence:
          - action: foxess_modbus.write_registers
            metadata: {}
            data:
              start_address: 40000
              values: >-
                {{now().strftime('%y')|int,now().strftime('%m')|int,now().strftime('%d')|int,now().strftime('%H')|int,now().strftime('%M')|int,now().strftime('%S')|int}}
              inverter: eebf190027c514026eba9631f55d1760
          - action: logbook.log
            metadata: {}
            data:
              name: RTCStoreInLog
              message: Updated time
mode: single
```

FozzieUK, 2025-02-09:
I think it’s because you’ve built a list of numbers which has an implicit start and end brackets (xx,xx,xx,xx,xx,xx,xx) whereas you want a string that looks like “xx,xx,xx,xx,xx,xx”

Try this instead
```
values: >-
    {{ now().strftime('%y')|int}},{{ now().strftime('%m')|int}},{{ now().strftime('%d')|int}},{{ now().strftime('%H')|int}},{{ now().strftime('%M')|int}},{{ now().strftime('%S')|int}}
```

WyndStryke, 2025-02-10:
Apologies for so many newbie questions, it works perfectly now, thank you :-)

No more worrying about clock drift.  I had a couple of my automations go wrong last night because they were relying on the inverter's _today figure being available near midnight, fortunately it was only data collection for my statistics rather than something functional.

WyndStryke, 2025-03-09:
FYI I think Fox *may* have fixed the time sync?  My automation has not had to do so many corrections recently.

FozzieUK, 2025-03-09:
> FYI I think Fox _may_ have fixed the time sync? My automation has not had to do so many corrections recently.

they fixed it not long after you got this working, and back to regular daily syncing again 👍 - be careful with your automation around the clock change as they will correct DST for you.

WyndStryke, 2025-03-09:
> be careful with your automation around the clock change as they will correct DST for you.

Yeah, I was worried about that, so I set it up so that if the difference is more than 5 minutes (or less than 2 seconds) it won't update, so when the clock change happens it should do nothing.

paulhargreaves, 2025-08-15:
https://github.com/nathanmarlor/foxess_modbus/issues/463#issuecomment-2646435061 doesn't appear to be "complete" - is there a complete (tested) version of this I can apply? I've disconnected mine from the internet so it's HA only, and would prefer to have time accurate so that it can run schedules properly.

WyndStryke, 2025-08-15:
Firstly I had limited understanding of how to write automations at the time I wrote it, and secondly the way that variable scoping works changed since that was written, so currently it is much more complex than it needs to be (should be a lot simpler if rewritten).

It works but it's not how I would suggest people do it now.

If the time is less than 2 seconds different it will leave it as-is.

If the time is more than 5 minutes different it will refuse to update.

It does not understand the clock change in spring and autumn.

```
alias: rtc_update_and_log
description: Write date & time from inverter into HA log
triggers:
  - trigger: time
    at: "23:30:30"
  - trigger: time
    at: "11:30:30"
conditions: []
actions:
  - variables:
      RTCRegister: null
  - action: foxess_modbus.read_registers
    data:
      start_address: 40000
      type: input
      inverter: eebf190027c514026eba9631f55d1760
      count: 6
    response_variable: RTCRegister
  - choose:
      - conditions:
          - condition: template
            value_template: >-
              {% set year = RTCRegister['values'].40000| int % 100 %} {% set
              month = RTCRegister['values'].40001| int %} {% set day =
              RTCRegister['values'].40002| int %} {% set hour =
              RTCRegister['values'].40003| int %} {% set minute =
              RTCRegister['values'].40004| int %} {% set second =
              RTCRegister['values'].40005| int %} {% set new_year =
              now().strftime('%y') | int   %} {% set new_month =
              now().strftime('%m') | int  %} {% set new_day =
              now().strftime('%d') | int  %} {% set new_hour =
              now().strftime('%H') | int  %} {% set new_minute =
              now().strftime('%M') | int  %} {% set new_second =
              now().strftime('%S') | int  %}   {% set safe_to_update = ( year ==
              new_year and month == new_month and hour == new_hour and
              (((minute-new_minute)|abs) < 5)) %} {% set need_to_update = not
              (year ==  new_year and month == new_month and hour == new_hour and
              minute == new_minute and ((second == new_second) or (((second -
              new_second) | abs) < 2))) %} {% set time_valid = not ((new_year <
              25) or (new_month < 1)  or (new_month > 12) or (new_day < 1) or
              (new_day > 31) or (new_hour < 0) or (new_hour > 23) or (new_minute
              < 0) or (new_minute > 59) or (new_second < 0)  or (new_second >
              59)) %} {{ need_to_update and safe_to_update and time_valid }}
        sequence:
          - action: foxess_modbus.write_registers
            metadata: {}
            data:
              start_address: 40000
              values: >-
                {{ now().strftime('%y')|int}},{{ now().strftime('%m')|int}},{{
                now().strftime('%d')|int}},{{ now().strftime('%H')|int}},{{
                now().strftime('%M')|int}},{{ now().strftime('%S')|int}}
              inverter: eebf190027c514026eba9631f55d1760
          - action: input_text.set_value
            metadata: {}
            data:
              value: >-
                {% set val = RTCRegister['error'] %} {% if 'Error reading
                registers' not in val %}
[... 58 lines cut ...]
            {{new_year}}/{{new_month}}/{{new_day}}
            {{new_hour}}:{{new_minute}}:{{new_second}}
              {%- if year == new_year and month == new_month and hour == new_hour and minute == new_minute and ((second == new_second) or (((second - new_second) | abs) < 2)) %}
            Time diff < 2 seconds, do not update.
              {%- else %}
            Inverter RTC time does not match HA time and needs to be updated.
              {%- endif %}
              {%- if year == new_year and month == new_month and hour == new_hour and (((minute-new_minute)|abs) < 5) %}
            Safe to update.
              {%- else %}
            Unsafe to update (time diff > 5 min).
              {%- endif %}
            {%- else %} No RTC Time, received {{ RTCRegister['error'] }} {%-
            endif %}
        target:
          entity_id: input_text.foxess_rtc_adjustments
      - action: logbook.log
        metadata: {}
        data:
          name: automation.rtc_update_and_log
          message: "{{states('input_text.foxess_rtc_adjustments')}}"
          entity_id: automation.rtc_update_and_log
mode: single

```