nathanmarlor/foxess_modbus #480: FrozenOrThawed metaclass in new HA breaks entity descriptions

canton7, 2023-12-23:
Looks like we'll need to do something like:

```py
class EntityFactoryMetaclass(type(EntityDescription), type(ABC)):
    pass

class ModbusBinarySensorDescription(BinarySensorEntityDescription, EntityFactory, metaclass=EntityFactoryMetaclass):
    ...
```

Alternatively, it looks like this might also work:

```py
class EntityFactory(metaclass=EntityFactoryMetaclass):
    ...
```

I'm not sure whethe rthis will work before EntityDescription's metaclass was introduced. Needs testing...

_Originally posted by @canton7 in https://github.com/nathanmarlor/foxess_modbus/discussions/479#discussioncomment-7933836_

FozzieUK, 2023-12-30:
@canton7 

Antony, my system has been working on the latest version since you modified the code, today I upgraded to the latest beta candidate and the Foxess_modbus integration fails to load with a metaclass warning.

I had another integration (Sun2) which was updated to support the new FrozenOrThawed metaclass and that has survived the upgrade.

I've tried a complete wipe, re-install and rebuilt it from scratch and confirmed it is the latest beta of HA Core which causes the problem.

Sadly it looks as if something wasn't stable on the dev release, i've downgraded back to it and Foxess_modbus works ok again - so seemingly there is still a breaking change in the current beta release candidate of core.

Previous (dev) ha core release - 2024.1.0.dev20231218

Latest (breaking) beta ha core release - 2024.1.0b2

I've included the logs below

```
2023-12-30 09:58:02.803 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing component custom_components.foxess_modbus
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/loader.py", line 822, in get_component
ComponentProtocol, importlib.import_module(self.pkg_path)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
return _bootstrap._gcd_import(name[level:], package, level)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
File "<frozen importlib._bootstrap_external>", line 940, in exec_module
File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
File "/config/custom_components/foxess_modbus/__init__.py", line 43, in <module>
from .inverter_profiles import inverter_connection_type_profile_from_config
File "/config/custom_components/foxess_modbus/inverter_profiles.py", line 25, in <module>
from .entities.charge_periods import CHARGE_PERIODS
File "/config/custom_components/foxess_modbus/entities/charge_periods.py", line 8, in <module>
from .modbus_charge_period_config import ChargePeriodAddressSpec
File "/config/custom_components/foxess_modbus/entities/modbus_charge_period_config.py", line 13, in <module>
from .modbus_binary_sensor import ModbusBinarySensorDescription
File "/config/custom_components/foxess_modbus/entities/modbus_binary_sensor.py", line 52, in <module>
class ModbusBinarySensor(ModbusEntityMixin, BinarySensorEntity):
TypeError: metaclass conflict: the metaclass of a derived class must be a (non-strict) subclass of the metaclasses of all its bases

2023-12-30 09:58:02.807 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration 'foxess_modbus': Unable to import component: Exception importing custom_components.foxess_modbus
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/loader.py", line 822, in get_component
ComponentProtocol, importlib.import_module(self.pkg_path)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
return _bootstrap._gcd_import(name[level:], package, level)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
File "<frozen importlib._bootstrap_external>", line 940, in exec_module
File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
File "/config/custom_components/foxess_modbus/__init__.py", line 43, in <module>
from .inverter_profiles import inverter_connection_type_profile_from_config
File "/config/custom_components/foxess_modbus/inverter_profiles.py", line 25, in <module>
from .entities.charge_periods import CHARGE_PERIODS
File "/config/custom_components/foxess_modbus/entities/charge_periods.py", line 8, in <module>
from .modbus_charge_period_config import ChargePeriodAddressSpec
File "/config/custom_components/foxess_modbus/entities/modbus_charge_period_config.py", line 13, in <module>
from .modbus_binary_sensor import ModbusBinarySensorDescription
File "/config/custom_components/foxess_modbus/entities/modbus_binary_sensor.py", line 52, in <module>
class ModbusBinarySensor(ModbusEntityMixin, BinarySensorEntity):
TypeError: metaclass conflict: the metaclass of a derived class must be a (non-strict) subclass of the metaclasses of all its bases
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/setup.py", line 251, in _async_setup_component
component = integration.get_component()
^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/loader.py", line 830, in get_component
raise ImportError(f"Exception importing {self.pkg_path}") from err
ImportError: Exception importing custom_components.foxess_modbus

```

canton7, 2023-12-30:
Interesting. I can't see any relevant PRs which have made it into HA, so I'm not sure what's happened on their side.

I also ran my little test script with the latest HA dev branch and that still passes.

I think I'll have to run up the latest docker image (they don't publish dev python packages as far as I can tell), but that'll have to wait a few days I'm afraid.

FozzieUK, 2023-12-30:
no problem, thanks 👍

canton7, 2024-01-02:
Aha, they went and added a metaclass to Entity as well.

canton7, 2024-01-02:
Annoyingly trying to do the same trick doesn't work. I'll have to come back to this one over the weekend

emontnemery, 2024-01-03:
FWIW, the core weather integration does like this to meet the metaclass requirements: 
https://github.com/joostlek/core/blob/0e807cd21315453d22ef79e892b5fceaddbe8b3e/homeassistant/components/weather/__init__.py#L264-L272

Ric57100, 2024-01-04:
Same issue since I updated HA from 2023.12.4 to 2024.1.0 with FoxEss Modbus v1.8.0.

Rollback to previous HA release.

Find below logs contents :

2024-01-04 08:51:34.432 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-04 08:51:34.470 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration liveboxtvuhd which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-04 08:51:34.474 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration husqvarna_automower which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-04 08:51:34.478 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration livebox which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-04 08:51:34.481 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration espsomfy_rts which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-04 08:51:34.486 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration foxess_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-04 08:51:34.490 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration xiaomi_cloud_map_extractor which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-04 08:52:02.622 WARNING (MainThread) [homeassistant.const] DATA_RATE_MEGABITS_PER_SECOND was used from livebox, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfDataRate.MEGABITS_PER_SECOND instead, please create a bug report at https://github.com/cyr-ius/hass-livebox-component/issues
2024-01-04 08:52:02.633 WARNING (MainThread) [homeassistant.components.sensor] STATE_CLASS_MEASUREMENT was used from livebox, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorStateClass.MEASUREMENT instead, please create a bug report at https://github.com/cyr-ius/hass-livebox-component/issues
2024-01-04 08:52:02.642 WARNING (MainThread) [homeassistant.components.sensor] STATE_CLASS_MEASUREMENT was used from livebox, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorStateClass.MEASUREMENT instead, please create a bug report at https://github.com/cyr-ius/hass-livebox-component/issues
2024-01-04 08:52:07.271 WARNING (MainThread) [py.warnings] /usr/local/lib/python3.11/site-packages/fuzzywuzzy/fuzz.py:11: UserWarning: Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning
  warnings.warn('Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning')

2024-01-04 08:52:09.846 WARNING (MainThread) [homeassistant.components.camera] SUPPORT_ON_OFF was used from xiaomi_cloud_map_extractor, this is a deprecated constant which will be removed in HA Core 2025.1. Use CameraEntityFeature.ON_OFF instead, please create a bug report at https://github.com/PiotrMachowski/Home-Assistant-custom-components-Xiaomi-Cloud-Map-Extractor/issues
2024-01-04 08:52:09.857 WARNING (MainThread) [homeassistant.components.camera] SUPPORT_ON_OFF was used from xiaomi_cloud_map_extractor, this is a deprecated constant which will be removed in HA Core 2025.1. Use CameraEntityFeature.ON_OFF instead, please create a bug report at https://github.com/PiotrMachowski/Home-Assistant-custom-components-Xiaomi-Cloud-Map-Extractor/issues
2024-01-04 08:52:11.867 WARNING (MainThread) [homeassistant.const] TIME_MINUTES was used from husqvarna_automower, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfTime.MINUTES instead, please create a bug report at https://github.com/Thomas55555/husqvarna_automower/issues
2024-01-04 08:52:12.866 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing component custom_components.foxess_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 822, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
[... cut ...] UREMENT: 'measurement'>, 'unit_of_measurement': 'W', 'device': {'identifiers': ['11'], 'manufacturer': 'LeChacal', 'model': 'rpict_11', 'name': 'RPICT 11', 'connections': []}, 'payload_not_available': 'offline', 'availability_mode': 'latest', 'payload_available': 'online', 'encoding': 'utf-8', 'enabled_by_default': True, 'qos': 0, 'force_update': False}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'ReactPower10'

calum-mcfarlane, 2024-01-04:
Presumably the thing to do is just not upgrade HA until this is resolved, right?

FozzieUK, 2024-01-04:
Yes definitely wait, it won’t be long and there are very likely other integrations that don’t work because of the changes made in HA core - keep a close eye on the issues area of your favourite integrations to make sure all are fixed before you upgrade.

muhkuh666, 2024-01-04:
Hi all
I have had an issue with the intergration after update to 2024.1.0 too.  When will there be a fix?
BR

canton7, 2024-01-04:
I was hoping to find time tonight but got stuck at work until late. Hopefully Saturday?

ppetru, 2024-01-06:
The underlying issue may or may not be the same, but for hass-weatherlink it was home-assistant/core#107149. A fix for that has been released in HA 2024.1.1.

muhkuh666, 2024-01-06:
> The underlying issue may or may not be the same, but for hass-weatherlink it was [home-assistant/core#107149](https://github.com/home-assistant/core/issues/107149). A fix for that has been released in HA 2024.1.1.

Did someone try?

martindell, 2024-01-06:
Nope :/ Still the same with 2024.1.1.  Reverting ..

`2024-01-06 08:38:41.719 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration foxess_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-01-06 08:40:05.871 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing component custom_components.foxess_modbus
File "/config/custom_components/foxess_modbus/__init__.py", line 43, in <module>
File "/config/custom_components/foxess_modbus/inverter_profiles.py", line 25, in <module>
File "/config/custom_components/foxess_modbus/entities/charge_periods.py", line 8, in <module>
File "/config/custom_components/foxess_modbus/entities/modbus_charge_period_config.py", line 13, in <module>
File "/config/custom_components/foxess_modbus/entities/modbus_binary_sensor.py", line 52, in <module>
2024-01-06 08:40:05.885 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration 'foxess_modbus': Unable to import component: Exception importing custom_components.foxess_modbus
File "/config/custom_components/foxess_modbus/__init__.py", line 43, in <module>
File "/config/custom_components/foxess_modbus/inverter_profiles.py", line 25, in <module>
File "/config/custom_components/foxess_modbus/entities/charge_periods.py", line 8, in <module>
File "/config/custom_components/foxess_modbus/entities/modbus_charge_period_config.py", line 13, in <module>
File "/config/custom_components/foxess_modbus/entities/modbus_binary_sensor.py", line 52, in <module>
ImportError: Exception importing custom_components.foxess_modbus`

vip-klaus, 2024-01-06:
yes, same problem with 2024.1.1. Only the weatherlink integration is working again with 2024.1.1.

FozzieUK, 2024-01-06:
The change in HA 2024.x needs changes in how affected integrations handle entities, the foxess-modbus integration work is in progress, please bear with us, once it is completed a new version of the integration will be released - hopefully not long now.

closed 2024-01-06 by canton7

muhkuh666, 2024-01-06:
Is the new Version available on HACS already?

soniiic, 2024-01-06:
> Is the new Version available on HACS already?

Yeah it's on HACS now

canton7, 2024-01-06:
This has been fixed in 1.8.1, which is available on HACS.

PR #484 (2023-12-24): Fix breakage on HA 2024.x
Fixes: #480
PR #487 (2024-01-06): Fix breakage caused by ABCCachedProperties metaclass on Entity
Fixes: #480
commit 0bb6ccef (2024-01-06): Fix breakage caused by ABCCachedProperties metaclass on Entity

Fixes: #480