nathanmarlor/foxess_modbus #495: FoxESS Integration Breaks After Update to HA Core 2024.1.0

Daz42, 2024-01-04:
Version v1.8.0 (latest)

FoxESS Integration Breaks After Update to HA Core 2024.1.0

I have tried upgrading HA Core a few times (rebooted/restored from backup and or snapshot - runs on VMware)

Logs:

Logger: homeassistant.setup
Source: setup.py:251
First occurred: 09:35:23 (1 occurrences)
Last logged: 09:35:23

Setup failed for custom integration 'foxess_modbus': Unable to import component: Exception importing custom_components.foxess_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 822, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/foxess_modbus/__init__.py", line 43, in <module>
    from .inverter_profiles import inverter_connection_type_profile_from_config
  File "/config/custom_components/foxess_modbus/inverter_profiles.py", line 25, in <module>
    from .entities.charge_periods import CHARGE_PERIODS
  File "/config/custom_components/foxess_modbus/entities/charge_periods.py", line 8, in <module>
    from .modbus_charge_period_config import ChargePeriodAddressSpec
  File "/config/custom_components/foxess_modbus/entities/modbus_charge_period_config.py", line 13, in <module>
    from .modbus_binary_sensor import ModbusBinarySensorDescription
  File "/config/custom_components/foxess_modbus/entities/modbus_binary_sensor.py", line 52, in <module>
    class ModbusBinarySensor(ModbusEntityMixin, BinarySensorEntity):
TypeError: metaclass conflict: the metaclass of a derived class must be a (non-strict) subclass of the metaclasses of all its bases

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 251, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 830, in get_component
    raise ImportError(f"Exception importing {self.pkg_path}") from err
ImportError: Exception importing custom_components.foxess_modbus

-------------------------------

Logger: homeassistant.loader
Source: loader.py:822
First occurred: 09:35:23 (1 occurrences)
Last logged: 09:35:23

Unexpected exception importing component custom_components.foxess_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 822, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/foxess_modbus/__init__.py", line 43, in <module>
    from .inverter_profiles import inverter_connection_type_profile_from_config
  File "/config/custom_components/foxess_modbus/inverter_profiles.py", line 25, in <module>
    from .entities.charge_periods import CHARGE_PERIODS
  File "/config/custom_components/foxess_modbus/entities/charge_periods.py", line 8, in <module>
    from .modbus_charge_period_config import ChargePeriodAddressSpec
  File "/config/custom_components/foxess_modbus/entities/modbus_charge_period_config.py", line 13, in <module>
    from .modbus_binary_sensor import ModbusBinarySensorDescription
  File "/config/custom_components/foxess_modbus/entities/modbus_binary_sensor.py", line 52, in <module>
    class ModbusBinarySensor(ModbusEntityMixin, BinarySensorEntity):
TypeError: metaclass conflict: the metaclass of a derived class must be a (non-strict) subclass of the metaclasses of all its bases

calum-mcfarlane, 2024-01-04:
Already logged here https://github.com/nathanmarlor/foxess_modbus/issues/480

darioz90, 2024-01-04:
Same problem after update 2024

vip-klaus, 2024-01-04:
Same problem here (2024.1)

canton7, 2024-01-04:
Duplicate of #480. **Please** people, follow the instructions in the issue template and look for pre-existing issues. It's pinned right at the top of the issue list.

closed 2024-01-04 by canton7