nathanmarlor/foxess_modbus #499: FoxESS Modbus not loading after upgrading to Home Assistant Core 2024.1

Grimroper, 2024-01-09:
Having upgraded Home Assistant Core from 2023.12.4 to 2024.1 the FoxESS Modbus integration is failing to load.  The logs report two issues with the Integration

1. Unexpected exception importing component custom_components.foxess_modbus
2. Setup failed for custom integration 'foxess_modbus': Unable to import component: Exception importing custom_components.foxess_modbus 

Rebooting/Restarting after upgrading to 2024.1 does not resolving the issue.  I subsequently upgrading Home Assistant to 2024.1.2 results in the same issue.  Reverting back to 2023.12.4 and the integration works as expected

```
Logger: homeassistant.loader
Source: loader.py:822
First occurred: 09:00:39 (1 occurrences)
Last logged: 09:00:39

Unexpected exception importing component custom_components.foxess_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 822, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/foxess_modbus/__init__.py", line 40, in <module>
    from .inverter_profiles import inverter_connection_type_profile_from_config
  File "/config/custom_components/foxess_modbus/inverter_profiles.py", line 21, in <module>
    from .entities.charge_periods import CHARGE_PERIODS
  File "/config/custom_components/foxess_modbus/entities/charge_periods.py", line 7, in <module>
    from .modbus_charge_period_config import ChargePeriodAddressSpec
  File "/config/custom_components/foxess_modbus/entities/modbus_charge_period_config.py", line 10, in <module>
    from .modbus_binary_sensor import ModbusBinarySensorDescription
  File "/config/custom_components/foxess_modbus/entities/modbus_binary_sensor.py", line 25, in <module>
    class ModbusBinarySensorDescription(BinarySensorEntityDescription, EntityFactory):
TypeError: metaclass conflict: the metaclass of a derived class must be a (non-strict) subclass of the metaclasses of all its bases
```

```
Logger: homeassistant.setup
Source: setup.py:251
First occurred: 09:00:39 (1 occurrences)
Last logged: 09:00:39

Setup failed for custom integration 'foxess_modbus': Unable to import component: Exception importing custom_components.foxess_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 822, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/foxess_modbus/__init__.py", line 40, in <module>
    from .inverter_profiles import inverter_connection_type_profile_from_config
  File "/config/custom_components/foxess_modbus/inverter_profiles.py", line 21, in <module>
    from .entities.charge_periods import CHARGE_PERIODS
  File "/config/custom_components/foxess_modbus/entities/charge_periods.py", line 7, in <module>
    from .modbus_charge_period_config import ChargePeriodAddressSpec
  File "/config/custom_components/foxess_modbus/entities/modbus_charge_period_config.py", line 10, in <module>
    from .modbus_binary_sensor import ModbusBinarySensorDescription
  File "/config/custom_components/foxess_modbus/entities/modbus_binary_sensor.py", line 25, in <module>
    class ModbusBinarySensorDescription(BinarySensorEntityDescription, EntityFactory):
TypeError: metaclass conflict: the metaclass of a derived class must be a (non-strict) subclass of the metaclasses of all its bases

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 251, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 830, in get_component
    raise ImportError(f"Exception importing {self.pkg_path}") from err
ImportError: Exception importing custom_components.foxess_modbus
```

canton7, 2024-01-09:
Duplicate of #480 (and #479, #490, #491, #492, #493, #494, #495). Fixed in v1.8.1.

The issue template says:

> Before you open a new issue, search through the existing issues to see if others have had the same problem

Please make sure you do this.

Thanks!

closed 2024-01-09 by canton7

Grimroper, 2024-01-09:
Sorry, didnt think to look in the close issues for a similar issue!!