nathanmarlor/foxess_modbus #506: Add support for the FoxESS KH 10.5 series, which currently appears to be unsupported.

canton7, 2024-01-13:
Add support for the FoxESS KH 10.5 series, which currently appears to be unsupported.

[image]

_Originally posted by @williamjeccles in https://github.com/nathanmarlor/foxess_modbus/discussions/505_

williamjeccles, 2024-01-13:
@canton7 I can now install the integration and link it to the USR-W610 but all the values are coming back unavailable
[image]

canton7, 2024-01-13:
Check the log (Settings - System - Logs). That normally means we're unable to talk to the adapter.

canton7, 2024-01-13:
I'm going to merge this. Feel free to hope back to #505 for help debugging your connection issues.

PR #507 (2024-01-13): Fix model detection for KH
It seems some models are just 'KHxx' not 'KH-xx'.

Fixes: #506
commit 0aa6786f (2024-01-13): Fix model detection for KH

It seems some models are just 'KHxx' not 'KH-xx'.

Fixes: #506
closed 2024-01-13 by canton7