nathanmarlor/foxess_modbus #509: Poll versions on (re-)connection

canton7, 2024-01-13:
It would be nice to get the versions of firmwares of the system as sensors in Homassistant.

The updates on my system are done by foxess or my electrician. So I do not know if a new firmware arrives. With the sensor it is possible to show and alert on change.

_Originally posted by @abuesen in https://github.com/nathanmarlor/foxess_modbus/discussions/504_

abuesen, 2024-02-22:
Thank you, inatelled beta: Looks all right, nice mdi!

I'm exited to see the first update.

lamixer, 2024-02-26:
Hello... KH10.5 user here. Running integration version 1.10.1 and I just noticed the three entities to display inverter firmware versions (like `sensor.foxess_manager_version`). I added these to a dashboard and they are there and say have been updated between 5 and 10 minutes ago, but the version numbers displayed do not match my inverter's real version numbers. For example, I'm running 1.19 of Manager but integration says I'm running 2.81. Not sure if I missed a memo or if that means it's not working on KH. Not a top priority! Thanks.

canton7, 2024-02-26:
@lamixer Thanks. Can you post your actual manager/master/slave versions, and the ones reported by this integration?

lamixer, 2024-02-26:
Sure... 

Actual:
Master 1.27 / Slave 1.01 / Manager 1.19

Integration:
Master: 2.95 / Slave 2.57 / Manager 2.81

abuesen, 2024-02-26:
I'm sorry I didn't remember the foxess-cloud at all since I use the great foxess-modbus integration. So my reported versions in other discussions are wrong. 

foxess-cloud says: Master : 1.83 Slave : 1.03 Manager : 1.66
foxess-modbus syas: Master : 1.83 Slave : 1.03 Manager : 3.85
Hardware: Master : 1.83 Slave : 1.03 Manager : 1.66

commit d4335d11 (2024-02-21): Read the version registers on (re-)connection

Fixes: #509
PR #565 (2024-03-16): The KH uses hex for its Manager version
Fixes: #509
commit c66a812e (2024-03-16): The KH uses hex for its Manager version

Fixes: #509
PR #568 (2024-03-17): Fix which versions are hex, again
Turns out that the KH has all hex, the H3 has just the manager version hex.

Fixes: #509
closed 2024-03-16 by canton7